hybridgroup/gobot #469: highgui.go: could not determine kind of name C.XXX

kpeu3i, 2017-11-22:
I try to compile this simple [example](https://gobot.io/documentation/platforms/opencv/) for ARM but get error:

```
# app/vendor/gocv.io/x/gocv
vendor/gocv.io/x/gocv/highgui.go:147:2: could not determine kind of name for C.Trackbar_Create
vendor/gocv.io/x/gocv/highgui.go:163:13: could not determine kind of name for C.Trackbar_GetPos
vendor/gocv.io/x/gocv/highgui.go:208:2: could not determine kind of name for C.Trackbar_SetMax
vendor/gocv.io/x/gocv/highgui.go:193:2: could not determine kind of name for C.Trackbar_SetMin
vendor/gocv.io/x/gocv/highgui.go:178:2: could not determine kind of name for C.Trackbar_SetPos
vendor/gocv.io/x/gocv/highgui.go:47:2: could not determine kind of name for C.Window_Close
vendor/gocv.io/x/gocv/highgui.go:114:2: could not determine kind of name for C.Window_IMShow
vendor/gocv.io/x/gocv/highgui.go:33:2: could not determine kind of name for C.Window_New
vendor/gocv.io/x/gocv/highgui.go:100:2: could not determine kind of name for C.Window_SetWindowProperty
vendor/gocv.io/x/gocv/highgui.go:126:13: could not determine kind of name for C.Window_WaitKey
```

Here is my Dockerfile:
```
FROM golang:1.9

MAINTAINER Andrey Bukatov <andrey.bukatov@gmail.com>

RUN echo "deb http://httpredir.debian.org/debian/ jessie main contrib non-free" >> /etc/apt/sources.list && \
    echo "deb http://httpredir.debian.org/debian/ jessie-updates main contrib non-free" >> /etc/apt/sources.list && \
    echo "deb http://security.debian.org jessie/updates main contrib non-free" >> /etc/apt/sources.list \
    echo "deb-src ftp://ftp.de.debian.org/debian/ stable main contrib" >> /etc/apt/sources.list

RUN apt-get update && apt-get install -y --no-install-recommends \
    gcc-arm-linux-gnueabi \
    g++-arm-linux-gnueabi \
    build-essential \
    zip unzip \
    pkg-config \
    cmake git libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev \
    python-dev python-numpy libtbb2 libtbb-dev libjpeg-dev libpng-dev libtiff-dev libjasper-dev libdc1394-22-dev

RUN mkdir -p /opencv && cd /opencv && \
	wget -O opencv.zip https://github.com/opencv/opencv/archive/3.3.1.zip && \
	wget -O opencv_contrib.zip https://github.com/opencv/opencv_contrib/archive/3.3.1.zip && \
	unzip opencv.zip && \
	unzip opencv_contrib.zip

ENV CC arm-linux-gnueabi-gcc
ENV CXX arm-linux-gnueabi-g++

RUN mkdir -p /opencv/opencv-3.3.1/build && cd /opencv/opencv-3.3.1/build && \
	cmake \
	    -D CMAKE_TOOLCHAIN_FILE=/opencv/opencv-3.3.1/platforms/linux/arm-gnueabi.toolchain.cmake \
	    -D CMAKE_BUILD_TYPE=RELEASE \
	    -D CMAKE_INSTALL_PREFIX=/usr/local \
	    -D OPENCV_EXTRA_MODULES_PATH=/opencv/opencv_contrib-3.3.1/modules \
        -D BUILD_DOCS=OFF \
        -D BUILD_EXAMPLES=OFF \
        -D BUILD_TESTS=OFF \
        -D BUILD_PERF_TESTS=OFF \
        -D BUILD_opencv_java=OFF \
        -D BUILD_opencv_python=OFF \
        -D BUILD_opencv_python2=OFF \
        -D BUILD_opencv_python3=OFF \
        ..

RUN sed -i 's/CMAKE_AR-NOTFOUND/arm-linux-gnueabi-ar/g' /opencv/opencv-3.3.1/build/CMakeFiles/3.7.2/CMakeCCompiler.cmake

RUN cd /opencv/opencv-3.3.1/build && \
	make -j8 && \
	make install && \
	ldconfig

ENV PKG_CONFIG_PATH /usr/local/lib/pkgconfig
ENV GOOS linux
ENV GOARCH arm
ENV GOARM 7
ENV CGO_ENABLED 1
ENV CGO_LDFLAGS -L/usr/local/lib -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_photo -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_dpm -lopencv_face -lopencv_fuzzy -lopencv_img_hash -lopencv_line_descriptor -lopencv_optflow -lopencv_reg -lopencv_rgbd -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_surface_matching -lopencv_tracking -lopencv_datasets -lopencv_text -lopencv_dnn -lopencv_plot -lopencv_xfeatures2d -lopencv_shape -lopencv_video -lopencv_ml -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_highgui -lopencv_videoio -lopencv_flann -lopencv_xobjdetect -lopencv_imgcodecs -lopencv_objdetect -lopencv_xphoto -lopencv_imgproc -lopencv_core
ENV CGO_CFLAGS -I/usr/local/include/opencv -I/usr/local/include
```
My compile command is as follow:
`docker run -it --rm -v "$PWD":/go -w /go/src/${PROJECT_NAME} opencv go build`

What I'm doing wrong?

deadprogram, 2017-11-22:
From the error, looks like perhaps you do not have the correct directory for the OpenCV `lib` files. I see you are creating your own Dockerfile. Are you sure that OpenCV is being built first before trying to build the GoCV code? I would expect it to take a long time just to build OpenCV itself.

kpeu3i, 2017-11-22:
Yes, of course. First of all, I build an image from a Dockerfile and only then compile GoCV code.
It might help (https://golang.org/cmd/cgo/#hdr-Go_references_to_C):
`To access a struct, union, or enum type directly, prefix it with struct_, union_, or enum_, as in C.struct_stat.`

deadprogram, 2017-11-22:
I think your linker settings are incorrect possibly? GoCV is known to work on Ubuntu, Arch Linux, OS X, and Windows 10.

I have not made any effort as of yet to get things working inside of a Docker container myself, but paging @trevrosen who knows more on this.

trevrosen, 2017-11-23:
(USA has the Thanksgiving holiday right now so I can’t give this a much attention until the weekend.)

I’ve been working on a Docker container that I’ll have up soon, but it’s mostly geared toward x86 / CI. Meanwhile I recommend using the build script for opencv in this repo’s root. I can run Gobot’s test suite in a container when I copy that script in to build opencv. 

Hope that helps.

kpeu3i, 2017-11-23:
Thanks for your reply! I would be thankful for any help.

My main goal is to build OpenCV for ARM architecture using Docker and then cross-compile GoCV code in a container.

deadprogram, 2017-11-23:
Hi @kpeu3i are you trying to use Gobot, or only GoCV? If the latter, seems like it would be best to open an issue in that repo https://github.com/hybridgroup/gocv

kpeu3i, 2017-11-23:
Hi!

I'm trying to run GoCV (face detection) on my robot (GoPiGo) which controlled by Gobot.

deadprogram, 2017-11-23:
OK, then you are in the right place. :+1: 

Also, sounds very cool. I want to do exactly the same thing.

trevrosen, 2017-11-28:
Hi @kpeu3i,
You won't be able to cross-compile GoCV, as it uses Cgo. You'll need to do an native compile of all components on ARM to achieve your goals.

Since the compilation of OpenCV works fine in this repo, I'm going to close this issue. Please feel free to open another one if you have trouble getting the existing OpenCV build scripts from this repo to work on ARM.

closed 2017-11-28 by trevrosen

kpeu3i, 2017-11-28:
Hi @trevrosen!
 
Could you please provide more information about the reason why I won't be able to cross-compile GoCV using CGO?

trevrosen, 2017-11-28:
@kpeu3i you can't cross compile **anything** that uses cgo. All cgo-implicating Go code must be compiled **on** the target architecture/OS.

kpeu3i, 2017-11-28:
Sorry, maybe I don't understand something, but I found this https://golang.org/cmd/cgo:

> When cross-compiling, you must specify a C cross-compiler for cgo to use. You can do this by setting the CC_FOR_TARGET environment variable when building the toolchain using make.bash, or by setting the CC environment variable any time you run the go tool. The CXX_FOR_TARGET and CXX environment variables work in a similar way for C++ code.`

trevrosen, 2017-11-28:
Apologies -- I should've been more specific:

When using cgo, you lose the `go build` command's support for making it "just work". In other words, the "effortless cross-compile" feature of Go is gone and you must manage every aspect of the cross compilation process yourself. This means not only setting things like `CC_FOR_TARGET` to working cross-compilers for each target platform, but also debugging any C-level issues that come up in the cross compilation process.

Unfortunately, helping you to achieve this goal is well outside the scope of the Gobot and GoCV projects at this time (though we'd be interested in **documentation** contributions for achieving cross compilation). I would suggest a workflow that involves maintaining parallel projects on your ARM and x86 machines and compiling on ARM to verify that something you've gotten working on x86 also works there.