hybridgroup/gobot #471: Question: motor driver for GoPiGo platform

kpeu3i, 2017-12-02:
Does Gobot support motor driver for GoPiGo platform?
I can't find any examples how to control motors. 

Thanks in advance!

deadprogram, 2017-12-02:
Hi @kpeu3i there is support for the GoPiGo3 motors:

https://godoc.org/gobot.io/x/gobot/platforms/dexter/gopigo3#Driver.SetMotorDps
https://godoc.org/gobot.io/x/gobot/platforms/dexter/gopigo3#Driver.SetMotorLimits
https://godoc.org/gobot.io/x/gobot/platforms/dexter/gopigo3#Driver.SetMotorPosition
https://godoc.org/gobot.io/x/gobot/platforms/dexter/gopigo3#Driver.SetMotorPower

There does not appear to be a specific example using those API, but we do have a few other examples to use as reference for the GoPiGo3 in general:

https://github.com/hybridgroup/gobot/blob/master/examples/gopigo3_servo.go

Hope that helps!

kpeu3i, 2017-12-02:
Hi, @deadprogram! You've helped me a lot! it works :)

Here is a simple working example (motors will rotate forth):

```
work := func() {
    gopigoDriver.SetMotorPower(gopigo3.MOTOR_LEFT+gopigo3.MOTOR_RIGHT, 100)
}

robot := gobot.NewRobot("gopigo",
    []gobot.Connection{raspiAdaptor, dualshockAdaptor},
    []gobot.Device{gopigoDriver, servoDriver, motorDriver},
    work,
)

robot.Start()
```

kpeu3i, 2017-12-02:
It looks like method `SetMotorDps` doesn't work:

```
work := func() {
    err := gopigoDriver.SetMotorDps(gopigo3.MOTOR_LEFT, 180)
    if err != nil {
        app.workError <- err
    }
}
```

Nothing happens when I run it on my robot. Did I miss something?

deadprogram, 2017-12-03:
I'm not completely sure what that function is supposed to do. Paging Dr. @ulisesflynn for assistance.

kpeu3i, 2017-12-03:
I think this function is a port of [GoPiGo Python Driver](https://github.com/DexterInd/GoPiGo3/blob/master/Software/Python/gopigo3.py#L480).

```
Set the motor target speed in degrees per second
    Keyword arguments:
        port -- The motor port(s). MOTOR_LEFT and/or MOTOR_RIGHT.
        dps -- The target speed in degrees per second
```

ulisesflynn, 2017-12-04:
@kpeu3i I'll look at this to see if there is a problem, thanks!

ulisesflynn, 2017-12-11:
@kpeu3i I found the problem and I will get a PR soon to fix that and a few other issues.

kpeu3i, 2017-12-11:
@ulisesflynn thanks a lot for your help in resolving the problem!

deadprogram, 2017-12-21:
This code was released in v1.8.0 so now closing. Please reopen if needed.

closed 2017-12-21 by deadprogram