hybridgroup/gobot #472: LED works, RGB not?

stuomas, 2017-12-06:
Hi, so I have one RGB led and I tried the basic example here https://gobot.io/documentation/platforms/raspi/ separately on all pins (11, 13, 15) and corresponding color fired up and blinks and everything is good.

Then I changed NewLedDriver to NewRgbLedDriver, keeping the physical connections the same. And my led does not respond to any of the functions. Anything else I need to know?

deadprogram, 2017-12-06:
Hello @stuomas it should work, can you please submit your exact code?

stuomas, 2017-12-06:
For example this works, and turns red led on.
`package main

import (
	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/raspi"
)

func main() {
	r := raspi.NewAdaptor()
	led := gpio.NewLedDriver(r, "11")

	work := func() {
		led.Off()
	}

	robot := gobot.NewRobot("ledbot",
		[]gobot.Connection{r},
		[]gobot.Device{led},
		work,
	)

	robot.Start()
}`

This doesn't respond to anything
`package main

import (
	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/raspi"
)

func main() {
	r := raspi.NewAdaptor()
	led := gpio.NewRgbLedDriver(r, "11", "13", "15")

	work := func() {
		led.SetRGB(255, 0, 0)
	}

	robot := gobot.NewRobot("ledbot",
		[]gobot.Connection{r},
		[]gobot.Device{led},
		work,
	)

	robot.Start()
}
`
On, Off, or Toggle neither work.

deadprogram, 2017-12-06:
So one important question @stuomas is have you installed pi-blaster in order to use PWM on the Raspi?

https://github.com/hybridgroup/gobot/tree/master/platforms/raspi#enabling-pwm-output-on-gpio-pins

You can now install pi-blaster using Gort, by install Gort on your Raspiberry Pi and then running the `gort raspi install piblaster` command.

stuomas, 2017-12-06:
@deadprogram I haven't. Is it mandatory with rgb functions even if you don't want to modulate the pins, but just on/off?

deadprogram, 2017-12-06:
Well, the assumption behind this particular driver is to have an LED that can support all of the RGB colors, which which requires PWM. That should probably be documented upfront.

That said, if https://github.com/hybridgroup/gobot/blob/master/drivers/gpio/rgb_led_driver.go#L141 could be modified to handle your use case.

For example:

```
if level == 0 || level == 255 {
    // use DigitalWriter instead of PwmWriter, cause we do not need it.
    return
}

// current logic using PwmWriter
```

What do you think?

stuomas, 2017-12-06:
I installed pi-blaster and now everything is fine. You're right it makes sense to require it, if you want to just turn on/off single colors of a rgb led, you can probably use the LedDriver for them separately. 

Maybe it could be mentioned in the documentation though :)

Thanks!

NomadMaker, 2018-06-08:
Do you need Pi-Blaster even if you use the native PWM pins of the Raspberry Pi? There are four hardware PWM pins that I hope will work without PI-Blaster.

Unfortunately Gort can't build pi-blaster for me. And manually fails in the same way.

deadprogram, 2018-06-08:
Hi @NomadMaker the PWM mapping of the Raspi Adaptor expects that pi-blaster is installed.

Can you please open a GH issue on Gort repo with details about your pi-blaster install issue?

lw664256297, 2019-12-25:
你好，抱歉打扰你，我在使用 NewRgbLedDriver 使用上面的代码，我的RGBLED不亮，我能确保我安装了 pi-blaster 并且我使用 NewLedDriver 能正常点亮
这是我的  pi-blaster 数据
MBox Board Revision: 0xc03111
DMA Channels Info: 0x71f5, using DMA Channel: 14
Using hardware:                   PWM
Number of channels:                32
PWM frequency:                 100 Hz
PWM steps:                       1000
Maximum period (100  %):      10000us
Minimum period (0.100%):         10us
DMA Base:                  0x20007000
Initialised, Daemonized, Reading /dev/pi-blaster.

[image]

closed 2024-02-21 by gen2thomas