hybridgroup/gobot #475: How to add work to an existent robot

usualwyy, 2017-12-21:
After a robot started,how to add work to the robot?

deadprogram, 2017-12-21:
Hello @usualwyy normally you would define all the possible work or commands, and then if you want to trigger additional behavior after the robot is started, you call those commands externally.

Does that match what you want to do?

usualwyy, 2017-12-21:
Thanks for fast response.
Yeah,I can do as what you said.But I think it's better to have a work pool so I can control the works dynamicly,may be in the future :)

nilskuhn, 2018-01-30:
@deadprogram Could you please show an example for defining commands that can be called after robot is started?

I'm controlling a stepper motor with the snippet below. But my robots work function never returns and hangs after printing _"Work done."_. My program acts like a daemon doing some house automization tasks on a raspberry pi and I would like to do the stepper motor (i.e. the robot) some work from time to time. What would be the way to go to achieve that?

```go
func (sdm *SanyoDenkiMotor) Rotate(revolutions float32, direction RotationDirection, speed RotationSpeed){

	motorWork := func() {

		defer sdm.deactivate()
		sdm.activate()

		stepsPerRevolution := int(sdm.GetBaseStepsPerRevolution()) * sdm.rMode.microStepCount
		var delay time.Duration
                delay = time.Duration(((1 / float32(stepsPerRevolution)) * float32(time.Second)) / float32(speed))
		steps := int(float32(stepsPerRevolution) * revolutions)

		sdm.SetDirection(direction)
		for i := 0; i < steps; i++ {
			sdm.stepPin.On()
			time.Sleep(delay * time.Nanosecond)
			sdm.stepPin.Off()
			time.Sleep(delay * time.Nanosecond)
		}
		fmt.Println("Work done.")
	}

	robot := gobot.NewRobot("SanyoDenkiStepper",
		[]gobot.Connection{sdm.rAdaptor},
		[]gobot.Device{sdm.dirPin, sdm.stepPin, sdm.sleepPin, sdm.modePins[0], sdm.modePins[1], sdm.modePins[2]},
		motorWork,
	)
	robot.Start()
}
```