hybridgroup/gobot #483: Mac OS: gocv compilation issue

sergeyu, 2018-01-24:
###  Steps:
- update homebrew
- install opencv (see details below)
- install gobot `go get -d -u gobot.io/x/gobot/...`
- tried to compile 'How to use'  openCV example

### Problem:

Got an error:
```
# gocv.io/x/gocv 
imgproc.cpp:61:28: error: expected '(' for function-style cast or type construction
```
Additional info:

```
opencv: stable 3.4.0 (bottled)
Open source computer vision library
https://opencv.org/
/usr/local/Cellar/opencv/3.4.0_1 (531 files, 97.7MB) *
  Poured from bottle on 2018-01-24 at 12:07:55
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/opencv.rb
```

```
go version go1.9.3 darwin/amd64
```

deadprogram, 2018-01-25:
Hi @sergeyu in order to use the GoCV support in Gobot, you will need to run the env.sh script that is included in GoCV to setup the needed environment vars: https://github.com/hybridgroup/gocv/blob/master/env.sh

This is needed in order to configure the CGo compilation settings for OpenCV.

deadprogram, 2018-02-14:
These docs have now been released as part of v1.9.0 thank you. Now closing.

closed 2018-02-14 by deadprogram