hybridgroup/gobot #484: How I can run gobot on microbit instantly without remote connection.

tttmaximttt, 2018-01-25:
Can I compile bin and upload it to microbit?

deadprogram, 2018-01-27:
Hello @tttmaximttt you can control a Microbit from a Go program running on a connected computer, but cannot compile/run Go on any RTOS yet. Seems like https://github.com/hybridgroup/gobot/blob/master/platforms/microbit/README.md could use a specific mention of this.

Hope that helps!

vendelin8, 2018-06-07:
Hi,

I had the same question. Is there a plan for this for the future? Is there anything I could help?
I'm asking it not specifically to Microbit, but in general, I'm more interested in Disigpark.

Thanks!

jadr2ddude, 2018-06-07:
@vendelin8 the main issue with this is that you cannot really run Go on a microcontroller. Most microcontrollers do not have enough flash to store the go runtime, plus it heavily relies on an OS. There is a project called emgo now, but it is far from usable at the moment - it doesnt even have memory management yet.