hybridgroup/gobot #485: Esp32 Dual Core

monouser7dig, 2018-01-25:
I‘d like to utilise the two cores of the esp32.
Is there any way to do this using go routines?

Thanks

deadprogram, 2018-01-27:
Hi @monouser7dig it is not yet possible to compile/run Go on any RTOS. I look forward to this capability someday!

monouser7dig, 2018-01-27:
I was thinking using gobot alone without an RTOS?
As far as I read gobot should be able to compile code for the esp32, only question is if some sort of concurrency can be used (easily). 

Or do you mean, using the two cores is not possible without an RTOS?
Is it possible to target each core using gobot so one could work on that?

deadprogram, 2018-01-27:
To clarify, there is not currently any way to execute Go code directly on any microcontrollers. The typical pattern for using microcontrollers from Gobot has been to run Go code on an embedded Linux, and communicate with a microcontroller using serial/wifi/bluetooth/i2c.

monouser7dig, 2018-01-27:
ok sorry, misunderstood, thought it was something along the lines of [micropython](https://micropython.org) providing a subset of go for micrcontrollers.

> Go on any RTOS. I look forward to this capability someday!

yes

closed 2018-01-27 by monouser7dig

deadprogram, 2018-01-27:
The llvm go compiler work can be found at http://llvm.org/viewvc/llvm-project/llvm/trunk/bindings/go/

Also check out this interesting project from @lpereira called Gomoku https://github.com/lpereira/gomoku

Lastly, this work also looks interesting: https://github.com/ycoroneos/G.E.R.T