hybridgroup/gobot #489: API Master command not in robeaux UI?

FerventGeek, 2018-02-07:
Using the example from the API overview page, I can add commands to both the Master gobot and Robots, but in the robeaux UI only the bot's commands are in the run drop-down. At the top-left, the master and the bots are listed and appear to be clickable, but nothing happens when toggling between them. Are these expected behaviors?

- The master's and robot's commands are in the http://{host}:3000/api/ JSON
- Tested on Chrome, FF and <shiver> Edge, same result

```golang
 gbot.AddCommand("custom_gobot_command",
    func(params map[string]interface{}) interface{} {
      return "This command is attached to the mcp!"
    })

  hello := gbot.AddRobot(gobot.NewRobot("hello"))

  hello.AddCommand("hi_there", func(params map[string]interface{}) interface{} {
    return fmt.Sprintf("This command is attached to the robot %v", hello.Name)
  })
```
Thx!

deadprogram, 2018-02-12:
Sounds like it might be an issue with Robeaux, not the api. We're gearing up for an iteration of improvements to the api and tooling soon...

FerventGeek, 2018-02-13:
Thanks. I'm showing this live on-stage for a Cisco IoT conference in May, but using a custom client for /API. The Robeaux interface is just sexy-cool to promote Gobot, so if it's not working, I can skip it. Thanks again, this is a really helpful library.