hybridgroup/gobot #491: Gobot Beaglebone Black GPIO question...

curtiswpaul, 2018-02-12:
I'm new to gobot...

I have a very simple Gobot/GPIO Button circuit.  
It should show output when button is pushed and then again when released, but it doesn't do anything.
It's not working.

Running image BBB-blank-debian-9.3-iot-armhf-2018-02-04-4gb.img.xz.  (I also tried the version posted on the BeagleBone site)
Modified /boot/uEnv.txt according to instructions at https://gobot.io/documentation/platforms/beaglebone.

I can see all the gpio's in /sys/class/gpio.

I'm using gpio67, so I cat the value file and see 0 when button is not pushed, then see 1 when button is pushed.  I know my circuit is good.

This is my gobot code for interacting with the button:
FYI...P8_8 is gpio67

package main

import (
        "fmt"

        "gobot.io/x/gobot"
        "gobot.io/x/gobot/drivers/gpio"
        "gobot.io/x/gobot/platforms/beaglebone"
)

func main() {
        beagleboneAdaptor := beaglebone.NewAdaptor()
        button := gpio.NewButtonDriver(beagleboneAdaptor, "P8_8")

        work := func() {
                button.On(gpio.ButtonPush, func(data interface{}) {
                        fmt.Println("button pressed")
                })

                button.On(gpio.ButtonRelease, func(data interface{}) {
                        fmt.Println("button released")
                })
        }

        robot := gobot.NewRobot("buttonBot",
                []gobot.Connection{beagleboneAdaptor},
                []gobot.Device{button},
                work,
        )

        robot.Start()
}

This is what it looks like when the code runs.  No errors, etc...
Push the button and don't get any of the Println's.

2018/02/12 04:14:03 Initializing connections...
2018/02/12 04:14:03 Initializing connection BeagleboneBlack-65FB3BAE ...
2018/02/12 04:14:03 Initializing devices...
2018/02/12 04:14:03 Initializing device Button-51E1127C ...
2018/02/12 04:14:03 Robot buttonBot initialized.
2018/02/12 04:14:03 Starting Robot buttonBot ...
2018/02/12 04:14:03 Starting connections...
2018/02/12 04:14:03 Starting connection BeagleboneBlack-65FB3BAE...
2018/02/12 04:14:03 Starting devices...
2018/02/12 04:14:03 Starting device Button-51E1127C on pin P8_8...
2018/02/12 04:14:03 Starting work...

Shouldn't I be seeing "button pressed" when I press the P8_8 button?....I don't see anything when I push the button.

deadprogram, 2018-02-12:
Hi @curtiswpaul sounds like a permissions problem, possibly. If you run your program under sudo, what happens?

curtiswpaul, 2018-02-12:
Same issue.  Also tried root.

deadprogram, 2018-02-12:
I'll pull out one of my BBBs, grab the latest OS image, and check it out, then.

curtiswpaul, 2018-02-12:
Got a recommendation on golang-nuts from Silviu Capota Mera to try something like this just to see if the code is reading the button push...hold down the button and execute code, etc...

Something like this...
dVal, err := beagleboneAdaptor.DigitalRead("P8_8")
if err != nil {
  fmt.Println("Error reading the pin: ", err)
  return
}
fmt.Println("Pin read correctly, the value is: ", dVal)

deadprogram, 2018-02-12:
Sure, that seems reasonable.

deadprogram, 2018-02-13:
OK @curtiswpaul I spent a few moments, and thanks to some help from @RobertCNelson I have some updates that will address this, especially to the initial setup instructions. I will get some updates committed on this later today.

curtiswpaul, 2018-02-13:
Right on!

deadprogram, 2018-02-13:
Hi @curtiswpaul I just pushed this commit to the `dev` branch: https://github.com/hybridgroup/gobot/commit/f54403303d7e5caca9c737358f0b8f44f9c60a7c

Please note the small difference in pin naming for a couple of the pins. The examples have all been updated to the correct names.

Let me know how this works for you.

curtiswpaul, 2018-02-14:
So the issue was the instructions for the uEnv.txt?
I changed button pin over to P8_10 and it works.

deadprogram, 2018-02-14:
There were 2 issues, one being the setup instructions, the other being that the pin naming changed at some point, requiring a leading zero in the file name used by the pin muxing. If you never used a single digit pin number, you would never notice, hence why we did not get a report until now.

In any case, I appreciate the issue report, and we will be releasing a new version soon.

deadprogram, 2018-02-14:
This code has now been released as part of v1.9.0 thank you again for your help. Now closing.

closed 2018-02-14 by deadprogram