hybridgroup/gobot #493: Is there any other way to install gobot?

ZLXT, 2018-02-22:
When i try to install gobot use the command "go get -d -u gobot.io/x/gobot/..." 。
My raspberry pi can't connect to the website。
Is there any other way to install it?
{
   package gobot.io/x/gobot/drivers/spi
        imports golang.org/x/exp/io/spi: unrecognized import path "golang.org/x/exp/io/spi" (https fetch: 
  Get https://golang.org/x/exp/io/spi?go-get=1: dial tcp 216.239.37.1:443: i/o timeout)
  package gobot.io/x/gobot/platforms/ble
          imports golang.org/x/sys/unix: unrecognized import path "golang.org/x/sys/unix" (https fetch: Get 
  https://golang.org/x/sys/unix?go-get=1: dial tcp 216.239.37.1:443: i/o timeout)
  package gobot.io/x/gobot/platforms/firmata
          imports go.bug.st/serial.v1: unrecognized import path "go.bug.st/serial.v1" (https fetch: Get 
 https://go.bug.st/serial.v1?go-get=1: dial tcp 5.39.77.130:443: i/o timeout)
}
GFW is really annoying.

deadprogram, 2018-02-22:
Hello @ZLXT if what you want is to execute a compiled Gobot program on your Raspi, you can cross compile it on another system, and then move the executable to the Raspi using your local network a USB thumb drive/etc. No need to install Go or Gobot on the Raspi itself, since Go produces a standalone executable program.

Hope that helps!

trevrosen, 2018-02-22:
@ZLXT just to expand on what @deadprogram said, keep in mind that Gobot is a library and there's not a runtime to "install". Because Go produces static binaries, your compiled Gobot program is all you need. Something like this:

`GOOS=linux GOARCH=arm7 go build` will produce an ARM Linux binary. Hope that helps!

closed 2018-02-22 by trevrosen