hybridgroup/gobot #497: Trouble with MCP3008 and Raspberry Pi

bloudermilk, 2018-02-27:
Hi there! I found this project the other day when looking for MCP3008 drivers in Go. I previously had my experiment wired up and successfully reading values from Python 3. However, I'm not having as much luck with Gobot (yet). I'm a complete Go novice and quite new to rpi/hardware in general, so it's probable that I'm making some beginner mistake. Any tips would be appreciated!

The following script is compiling and running on the pi just fine, but I only get zeros out.

```go
package main

import (
        "time"
        "fmt"

        "gobot.io/x/gobot"
        "gobot.io/x/gobot/drivers/spi"
        "gobot.io/x/gobot/platforms/raspi"
)

func main() {
        raspi := raspi.NewAdaptor()
        adc := spi.NewMCP3008Driver(raspi)

        work := func() {
                adc.Start()

                gobot.Every(1*time.Second, func() {
                        result, err := adc.Read(0)

                        if err == nil {
                          fmt.Printf("Hello world: %v\n", result)
                        } else {
                          fmt.Printf("Error\n")
                        }
                })
        }

        robot := gobot.NewRobot("logger",
                []gobot.Connection{raspi},
                []gobot.Device{adc},
                work,
        )

        robot.Start()
}
```

This is my Python script which works just fine:

```python
# simple example of reading the mcp3008 analog input channels and printing
# them all out.
# author: tony dicola
# license: public domain
import time

# import spi library (for hardware spi) and mcp3008 library.
import adafruit_gpio.spi as spi
import adafruit_mcp3008

# hardware spi configuration:
spi_port   = 0
spi_device = 1
mcp = adafruit_mcp3008.mcp3008(spi=spi.spidev(spi_port, spi_device))

print('reading mcp3008 values, press ctrl-c to quit...')
# print nice channel column headers.
print('| {0:>4} | {1:>4} | {2:>4} | {3:>4} | {4:>4} | {5:>4} | {6:>4} | {7:>4} |'.format(*range(8)))
print('-' * 57)
# main program loop.
while true:
    # read all the adc channel values in a list.
    values = [0]*8
    for i in range(8):
        # the read_adc function will get the value of the specified channel (0-7).
        values[i] = mcp.read_adc(i)
    # print the adc values.
    print('| {0:>4} | {1:>4} | {2:>4} | {3:>4} | {4:>4} | {5:>4} | {6:>4} | {7:>4} |'.format(*values))
    # pause for half a second.
    time.sleep(0.5)
```

deadprogram, 2018-03-27:
Hi @bloudermilk sorry about the delayed response, somehow just noticed this.

I pulled out my MCP3008 (the RasPiO Analog Zero) and just need to solder it up to look into this.

However, I have to travel this week, so I will probably not be able to help until next week.

deadprogram, 2018-03-27:
One thing I just noticed, you should not call `adc.Start()` in your `work` function. it is called for you by `robot.Start()`.

deadprogram, 2018-04-10:
There is a problem with reading SPI data that I am currently working on. 

There is a very high probability of Gobot changing out the SPI API, which will fix the devices, but break the API for device implementors.

More soon...

bloudermilk, 2018-04-10:
@deadprogram thank you for following up. Now I must offer you an apology for going MIA on this issue. This week I will try to recompile my program without `adc.Start` to see if that has any affect. Let me know if there's anything else I can do to help you debug this. Happy to lend my Pi over SSH if that's helpful.

deadprogram, 2018-04-12:
Hi @bloudermilk the `dev` branch has a new set of code for SPI devices that should work a lot better for you, and others.

The sample program is located here: https://github.com/hybridgroup/gobot/blob/dev/examples/raspi_mcp3008.go

Please let me know if that solves your issue for your particular configuration.

deadprogram, 2018-04-20:
This has now been released as part of v1.10.0 so I am now closing this issue. Please reopen if needed, @bloudermilk and thanks.

closed 2018-04-20 by deadprogram