hybridgroup/gobot #498: Add custom route in api.go?

FerventGeek, 2018-03-06:
I might be missing it, but can I add custom routes in API? For example, rather than open the default `/` with the Robeaux `index.html`, open a custom index page with a bot-specific custom UI, using `/API` underneath to make magic happen.  I can hack the source to refactor the current HTML into another page, but that's clumsy, and I heard a rumor you're making changes to the Robeaux/API features anyway.

FerventGeek, 2018-03-08:
Update. Close this :) I realized you can override any existing route after the API initializer is called.

```golang
a := api.NewAPI(gbot)
a.Get("/", func(res http.ResponseWriter, req *http.Request) {
		res.Write([]byte("hey there i see you!"))
	})
```

I'm good!  Thx

bloudermilk, 2018-03-09:
@FerventGeek you can actually close this issue yourself since you opened it. You should see a 'Close issue' button next to the 'Comment' button. 🍻

closed 2018-03-09 by trevrosen