hybridgroup/gobot #501: Mifare MFRC522 RFID Reader/Writer support?

simonbrugger, 2018-03-23:
Hi, 

do gobot support the Mifare MFRC522 RFID Reader/Writer in some way? I saw that there are already some SPI drivers available but not for MFRC522. Is there any default way to communicate with one SPI device? 

And what is the best way to contribute a MFRC522 driver?

deadprogram, 2018-04-10:
There is a problem with reading SPI data that we are currently working on. 

That means there is a very high probability of Gobot changing out the SPI API, which will fix the devices, but break the API for device implementors.

Once this work is completed, we will update the code generator to assist in creating the skeleton for new SPI drivers.

deadprogram, 2018-04-12:
Hi @simonbrugger the `dev` branch now has the new SPI implementation.

Take a look at any of the SPI driver implementations for inspiration. Seems like the MFRC522 has a complex API of its own, so there appears to be a fair bit of work needed to support all of its features.

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas