hybridgroup/gobot #502: Scanner error

ghostsquad, 2018-03-26:
```
$ go run ./platforms/joystick/bin/scanner.go
# command-line-arguments
platforms/joystick/bin/scanner.go:45:50: cannot use t.Which (type sdl.JoystickID) as type int in argument to sdl.JoystickOpen
```

Not sure what to do about this error. I have a 360 controller plugged in, and verified that it's detected and working through this:
https://github.com/360Controller/360Controller

deadprogram, 2018-03-26:
Looks like either a bug or an API change. Either way, needs to be looked into.

deadprogram, 2018-03-31:
Hi @ghostsquad I just committed this branch with what should address this issue: https://github.com/hybridgroup/gobot/tree/bugfix/joystick-scanner

If you have a chance, can you please give it a try? I am currently travelling and do not have any joystick/gamepad to try out.

ghostsquad, 2018-03-31:
I'll try it out. Stay tuned.

deadprogram, 2018-04-12:
I've added code and merged into the `dev` branch, which is working on my machine:

```
$ ./scanner 
Joystick 0 connected
[5880 ms] Axis: 4       value:-32768
[5880 ms] Axis: 5       value:-32768
```

I will leave this issue open until the next release.

ghostsquad, 2018-04-13:
Sorry I haven't had a chance to check this out. I'm terrible.

gherlein, 2018-04-14:
this branch fixes the error for me, thanks.

deadprogram, 2018-04-20:
This has now been released as part of v1.10.0 so now closing this issue. Thank you everyone!

closed 2018-04-20 by deadprogram