hybridgroup/gobot #504: Support for Olimex LIME2?

eric-s-raymond, 2018-03-27:
The Olimex LIME2 is a Linux SBC based on the Allwinner A20 SOC. It's not on your platform list. I just found Gobot today  and the advantage of being able to use it looks a lot bigger than the 80 Euros I paid for this hardware.

The result is that I'm probably going to drop the LIME2 for my current project and use a BeagleBone Black instead.  (Can't use an RPi, lack of USB gadget port is a blocker.)

Consequently, I now have a LIME2 sitting idle.  II'll mail it to you if you undertake to do the port; that way at least somebody wins. I would be astonished if the port were in the least difficult; it seems to run a pretty stock Debian.

esr@thyrsus,com if you prefer email.

gen2thomas, 2024-02-21:
https://www.olimex.com/Products/OLinuXino/A20/A20-OLinuXino-LIME2/open-source-hardware

gen2thomas, 2025-03-21:
https://www.armbian.com/olimex-lime-2/