hybridgroup/gobot #509: Segmentation Fault when using SPI with Beaglebone Black

bodokaiser, 2018-04-07:
I am working on a driver for the AD9910 direct-digital-synthesizer.

```go
package driver

import (
	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/spi"

	"github.com/bodokaiser/beagle/device/register"
)

// AD9910 communication ops.
const (
	AD9910Write byte = 0x80
	AD9910Read       = 0x00
)

// AD9910 register addresses.
const (
	AD9910CtrlFunc1 = iota
	AD9910CtrlFunc2
	AD9910CtrlFunc3
	AD9910AuxDACCtrl
	AD9910IOUpdateRate
	_
	_
	AD9910FreqTuningWord
	AD9910PhaseOffsetWord
	AD9910AmplScaleFactor
	AD9910MultichipSync
	AD9910DigitalRampLimit
	AD9910DigitalRampStepSize
	AD9910DigitalRampRate
	AD9910Profile0
	AD9910Profile1
	AD9910Profile2
	AD9910Profile3
	AD9910Profile4
	AD9910Profile5
	AD9910Profile6
	AD9910Profile7
	AD9910RAM
)

// AD9910 is the gobot driver for the AD9910 DDS.
type AD9910 struct {
	name       string
	connector  spi.Connector
	connection spi.Connection
	registers  *register.AD9910
}

// NewAD9910 returns an initialized AD9910.
func NewAD9910(c spi.Connector) *AD9910 {
	r := new(register.AD9910)
	r.CtrlFunc1Data = register.AD9910CtrlFunc1Default
	r.CtrlFunc2Data = register.AD9910CtrlFunc2Default
	r.CtrlFunc3Data = register.AD9910CtrlFunc3Default
	r.AuxDACCtrlData = register.AD9910AuxDACCtrlDefault
	r.StProfile1Data = register.AD9910StProfile0Default

	return &AD9910{
		name:      gobot.DefaultName("AD9910"),
		connector: c,
		registers: r,
	}
}

// Name returns the name of the device.
func (d *AD9910) Name() string {
	return d.name
}

// SetName sets the name of the device.
func (d *AD9910) SetName(s string) {
	d.name = s
}

// Connection returns the spi connection used by the device.
func (d *AD9910) Connection() gobot.Connection {
	return d.connection.(gobot.Connection)
}

// RunSingleTone configures AD9910 to run in single tone mode with amplite,
// frequency and phase.
func (d *AD9910) RunSingleTone(ampl uint16, freq uint32, phase uint16) error {
	d.registers.SetRAMEnable(false)
	d.registers.SetDigitalRampEnable(false)
	d.registers.SetParallelPortEnable(false)

	d.registers.SetStProfile0AmplScaleFactor(ampl)
	d.registers.SetStProfile0FreqTuningWord(freq)
	d.registers.SetStProfile0PhaseOffsetWord(phase)

	instructions := [][]byte{
		append([]byte{AD9910CtrlFunc1 | AD9910Write}, d.registers.CtrlFunc1Data[:]...),
		append([]byte{AD9910CtrlFunc2 | AD9910Write}, d.registers.CtrlFunc2Data[:]...),
		append([]byte{AD9910CtrlFunc3 | AD9910Write}, d.registers.CtrlFunc3Data[:]...),
	}

	w := []byte{}
	for _, i := range instructions {
		w = append(w, i...)
	}
	r := make([]byte, len(w))

	return d.connection.Tx(w, r)
}

// Start initializes the driver.
func (d *AD9910) Start() (err error) {
	bus := d.connector.GetSpiDefaultBus()
	mode := d.connector.GetSpiDefaultMode()
	speed := d.connector.GetSpiDefaultMaxSpeed()

	d.connection, err = d.connector.GetSpiConnection(bus, mode, speed)

	return err
}

// Halt stops the driver.
func (d *AD9910) Halt() (err error) {
	return d.connection.Close()
}
```

and running the whole thing with

```go
func main() {
	beagle := beaglebone.NewAdaptor()

	dds0 := driver.NewAD9910(beagle)
	dds0.Start()
	dds0.RunSingleTone(1<<15, 1<<30, 0)
}
```

however I am experiencing a segmentation fault when doing the spi transaction `d.connection.Tx(...)`.

I am using the official Debian stretch 9.4 image flashed to MMC. The universal cape manager is enabled. Using [adafruits BeagleBone python io package](https://github.com/adafruit/adafruit-beaglebone-io-python) I am able to successfully use the SPI interface, thus I don't think this is a hardware / os configuration problem.

deadprogram, 2018-04-10:
There is a problem with reading SPI data that I am currently working on.

There is a very high probability of Gobot changing out the SPI API, which will fix the devices, but break the API for device implementors.

One of the needed changes would accommodate setting both the device bus ID, and chip ID. This would address your particular concerns, @bodokaiser 

More soon...

deadprogram, 2018-04-12:
Hi @bodokaiser please take a look at the `dev` branch which has the new SPI implementation.

All the the spi drivers now support optional params. You can now set the specific device bus and chip like this:

```go
a := raspi.NewAdaptor()
adc := spi.NewMCP3008Driver(a, spi.WithBus(1), spi.WithChip(0))
```

Which should give you access to `/dev/spidev1.0` as you require.

Please give it a try, your feedback is appreciated.

bodokaiser, 2018-04-12:
I can confirm that 

```go
package main

import (
	"log"

	"gobot.io/x/gobot/platforms/beaglebone"
)

func main() {
	beagle := beaglebone.NewAdaptor()

	c, err := beagle.GetSpiConnection(1, 0, 0, 8, 5e6)
	if err != nil {
		log.Fatal(err)
	}

	w := []byte{
		8, 0, 0,
		1, 3, 64, 8, 32,
		14, 63, 255, 0, 0, 1, 137, 0, 0,
		0, 0, 128, 2, 2,
		9, 0, 0, 255, 252,
		7, 1, 137, 0, 0,
	}
	r := make([]byte, len(w))

	err = c.Tx(w, r)
	if err != nil {
		log.Fatal(err)
	}

	log.Printf("%v", r)
}
```

works as expected. Thank you!