hybridgroup/gobot #515: Tello and video on Windows 10 non start

fabriziomarini, 2018-04-29:
I'm using gobot with tello and it is ok .... but if i try tello_video.go with Mplayer installed on Windows 10 I only see in the output:
2018/04/29 18:24:43 Initializing connections...
2018/04/29 18:24:43 Initializing devices...
2018/04/29 18:24:43 Initializing device Tello-6DD8127A03A3B7AB ...
2018/04/29 18:24:43 Robot tello initialized.
2018/04/29 18:24:43 Starting Robot tello ...
2018/04/29 18:24:43 Starting connections...
2018/04/29 18:24:43 Starting devices...
2018/04/29 18:24:43 Starting device Tello-6DD8127A03A3B7AB...
2018/04/29 18:24:43 Starting work...

No video start ... tello it is connected

deadprogram, 2018-04-30:
Hi @fabriziomarini just wondering if you need to change the line https://github.com/hybridgroup/gobot/blob/master/examples/tello_video.go#L29 to use `"mplayer.exe"` as the first param to the `exec.Command()` function due to running on Windows. Please give that a try, and let me know.

fabriziomarini, 2018-04-30:
I have tried but nothing change.
I have also used a cmd as administrator to start the "go run ...."
My dubt is that I do no see "Connected" on the console the one used in line https://github.com/hybridgroup/gobot/blob/master/examples/tello_video.go#L37

deadprogram, 2018-04-30:
Are you able to run the example `tello.go`?

fabriziomarini, 2018-04-30:
Yes tello.go work fine.

deadprogram, 2018-04-30:
What version of Windows are you using? Also, is mplayer.exe in your `PATH`?

deadprogram, 2018-04-30:
Also, I wonder if https://www.reddit.com/r/golang/comments/8e9las/problem_using_readfromudp_on_windows_to_discover/ might be related?

fabriziomarini, 2018-04-30:
OS Name	Microsoft Windows 10 Pro
Version	10.0.15063 Build 15063
Yes mplayer it is in the path

gen2thomas, 2023-07-07:
Hi @fabriziomarini and @deadprogram , because there seems to be no related issue since 2018 for MS Windows, my suggestion is to close this issue.

deadprogram, 2023-07-07:
Agreed. Thanks everyone!

closed 2023-07-07 by deadprogram