hybridgroup/gobot #516: Test failures on 32 bits platform

maruel, 2018-05-01:
On an ODROID-C1+ running Ubuntu 16.04, I'm now able to run almost all tests;

First with some prerequisites:
```
sudo apt install libusb-dev libsdl2{,-image,-mixer,-ttf,-gfx}-dev
```

Then running the tests:
```
 ~/go/src/gobot.io/x/gobot$ ./xvfb.sh go test ./...
(successes elided)
# gobot.io/x/gobot/platforms/joystick/bin
platforms/joystick/bin/scanner.go:45:50: cannot use t.Which (type sdl.JoystickID) as type int in argument to sdl.JoystickOpen
# gobot.io/x/gobot/platforms/leap
platforms/leap/leap_motion_driver_test.go:92:45: constant 4729292670 overflows int
FAIL    gobot.io/x/gobot/platforms/leap [build failed]
--- FAIL: TestMqttAdaptorConnectWithAuthError (0.00s)
        gobottest.go:14: mqtt_adaptor_test.go:93: 1 error occurred:

                * Network Error : dial tcp [::1]:1883: connect: connection refused - "*multierror.Error", should equal,  1 error occurred:

                * Network Error : Unknown protocol - "*multierror.Error"
FAIL
FAIL    gobot.io/x/gobot/platforms/mqtt 0.038s
```

- platforms/leap/leap_motion_driver_test.go seems easy to fix, the test just fails on 32 bits platform.
- platforms/joystick/bin/scanner.go is probably related to the fact I installed an old sdl2 version (2.0.4), which prints a ton of warnings. Haven't investigated yet.
- platforms/mqtt/mqtt_adaptor_test.go is kinda weird, not sure why this test exists.

Where xvfb.sh is:
```
#/bin/sh
# Run under xvfb

set -eu

if (which xvfb-run>/dev/null); then
  xvfb-run \
    -a "--server-args=-screen 0 1280x800x24 -ac -nolisten tcp -dpi 98 +extension RANDR" \
    "$@"
else
  echo "Please 'sudo apt-get install xvfb' first"
  exit 1
fi
```

jadr2ddude, 2018-05-17:
For the platforms/leap/leap_motion_driver_test.go fail:

Please see [this line](https://github.com/hybridgroup/gobot/blob/b2ddfe7895c4e414cd4e6a6734b31b12ea68ed8e/platforms/leap/parser.go#L69) for one of the main issues - the int is 32-bits on 32-bit systems, and thus overflows. The frame should be changed to either uint64 or int64 to maintain an appropriate width on 32-bit platforms. Also we will need to specify the number type for the test value [here](https://github.com/hybridgroup/gobot/blob/b2ddfe7895c4e414cd4e6a6734b31b12ea68ed8e/platforms/leap/leap_motion_driver_test.go#L92).

jadr2ddude, 2018-05-18:
should i submit a PR?

deadprogram, 2018-05-18:
Yes, please, and thank you!

jadr2ddude, 2018-05-19:
@deadprogram how do the MQTT tests work? is there some hidden internal server that they are testing against?

jadr2ddude, 2018-05-19:
Ok i think i have deciphered it. I think paho changed to return the underlying connect error, and this code has not been updated.

jadr2ddude, 2018-05-19:
I just ran the SDL build on my machine and I got the same error (using Manjaro - pretty recent). I will investigate this.

jadr2ddude, 2018-05-19:
SDL was a pretty quick fix. Sending a PR.

commit 10789b99 (2018-05-19): change timestamp to uint64 to fix #516
commit 108cfecd (2018-05-19): Merge pull request #540 from jadr2ddude/dev

change timestamp to uint64 to fix #516