hybridgroup/gobot #517: I2C LCD Driver Example Program

adityaalviori, 2018-05-02:
Hello All, 
I am newbie on Golang and used BeagleBone Board. 

May I request the example how to program the LCD using I2C, from how to connect the BB to LCD then writing the string on LCD

Thanks for your help

gen2thomas, 2022-08-25:
Hi, although the question is quite old I will try to answer it to allow closing the Issue.

Meanwhile there is a driver and an example available for Tinkerboard: https://github.com/hybridgroup/gobot/blob/release/examples/tinkerboard_adafruit1109_lcd_keys.go

Possibly someone with a BB can adapt it to BB and add it to the repository.

Hope this helps.

gen2thomas, 2022-10-19:
I'm going to close this issue now. If a special device is needed, please open a device request issue.

closed 2022-10-19 by gen2thomas