hybridgroup/gobot #518: Unable to start video on ubuntu 16.04

rlanari, 2018-05-03:
Hi, I'm trying to run your example (go run tello_video.go): all seems to work (tello connection is ok.. the last message displayed is "connected"), but mplayer does not displays any video.
The command is correctly run (ps -ef | grep mplayer --> i find it as a running process with -fps 25 -)... but nothing is displayed.
DO you have any suggestiong about it ?
Many thanks in advance and congratulation for your hard work !!!

deadprogram, 2018-05-03:
Hi @rlanari does the player window open, but no video in it? 

I don;t think you should find the mplayer process running separately from the go process when the go code is not running. Perhaps you need to kill a stray mplayer process?

rlanari, 2018-05-03:
I checked it but before running go process I didn't find any mplayer process: I saw mplayer process only after typing the command: go run tello_video.go
[image]
 I have this OS:
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.4 LTS"

What OS are you using ?

deadprogram, 2018-05-03:
I am using same Ubuntu as you, and I installed from package manager. So nothing very interesting.

What output do you get when you run `mplayer -h`?

deadprogram, 2018-05-03:
Also, what version of Go?

rlanari, 2018-05-04:
mplayer -h
Usage:   mplayer [options] [url|path/]filename

Basic options: (complete list in the man page)
 -vo <drv>        select video output driver ('-vo help' for a list)
 -ao <drv>        select audio output driver ('-ao help' for a list)
 vcd://<trackno>  play (S)VCD (Super Video CD) track (raw device, no mount)
 dvd://<titleno>  play DVD title from device instead of plain file
 -alang/-slang    select DVD audio/subtitle language (by 2-char country code)
 -ss <position>   seek to given (seconds or hh:mm:ss) position
 -nosound         do not play sound
 -fs              fullscreen playback (or -vm, -zoom, details in the man page)
 -x <x> -y <y>    set display resolution (for use with -vm or -zoom)
 -sub <file>      specify subtitle file to use (also see -subfps, -subdelay)
 -playlist <file> specify playlist file
 -vid x -aid y    select video (x) and audio (y) stream to play
 -fps x -srate y  change video (x fps) and audio (y Hz) rate
 -pp <quality>    enable postprocessing filter (details in the man page)
 -framedrop       enable frame dropping (for slow machines)

Basic keys: (complete list in the man page, also check input.conf)
 <-  or  ->       seek backward/forward 10 seconds
 down or up       seek backward/forward  1 minute
 pgdown or pgup   seek backward/forward 10 minutes
 < or >           step backward/forward in playlist
 p or SPACE       pause movie (press any key to continue)
 q or ESC         stop playing and quit program
 + or -           adjust audio delay by +/- 0.1 second
 o                cycle OSD mode:  none / seekbar / seekbar + timer
 * or /           increase or decrease PCM volume
 x or z           adjust subtitle delay by +/- 0.1 second
 r or t           adjust subtitle position up/down, also see -vf expand

 * * * SEE THE MAN PAGE FOR DETAILS, FURTHER (ADVANCED) OPTIONS AND KEYS * * *

MPlayer 1.2.1 (Debian), built with gcc-5.3.1 (C) 2000-2016 MPlayer Team

rlanari, 2018-05-04:
go version
go version go1.10.2 linux/amd64

deadprogram, 2018-05-04:
You are using pretty much the exact same versions of everything as I am.

OK one thought is are you already using port 6038 for something else? I have intentions to allow changing the port used for the video when received, but still have not implemented that: https://github.com/hybridgroup/gobot/blob/master/platforms/dji/tello/driver.go#L248

jyzadeh, 2018-05-06:
Hi @rlanari, I'm having the same issue. Have you found a fix?

NikolaeVarius, 2018-08-17:
I can't run this with arch either. The tello works fine but no video window opens

sebsto, 2018-08-26:
I had a similar issue on MacOSX and found out that my company IT enforced a firewall rule on our laptop preventing to accept UDP packets. Disabling the firewall solved the problem. 

On Mac : type sudo pfctl -d