hybridgroup/gobot #519: Tello OpenCV with ffmpeg

JakeYoussefzadeh, 2018-05-04:
I have an issue running the gobot/examples/tello_opencv.go example. 

I have ffmpeg and opencv installed, and have run the env.sh script before running the opencv file. 

When running the file, the output is flooded with "EOF", and the occasional "write |1: broken pipe".

Here is the initial output of running env.sh and tello_opencv.go:

```
Jakes-MacBook-Pro-5:~ jake$ source go/src/gocv.io/x/gocv/env.sh
Brew install detected
Environment variables configured for OSX
Jakes-MacBook-Pro-5:~ jake$ go run /Users/jake/Desktop/cv.go 
2018/05/04 12:02:02 Initializing connections...
2018/05/04 12:02:02 Initializing devices...
2018/05/04 12:02:02 Initializing device Tello-7270070BB4EEB5CD ...
2018/05/04 12:02:02 Initializing device Window ...
2018/05/04 12:02:02 Robot tello initialized.
2018/05/04 12:02:02 Starting Robot tello ...
2018/05/04 12:02:02 Starting connections...
2018/05/04 12:02:02 Starting devices...
2018/05/04 12:02:02 Starting device Tello-7270070BB4EEB5CD...
2018/05/04 12:02:02 Starting device Window...
2018/05/04 12:02:02 Starting work...
EOF
...
```

Furthermore, whenever attempting to edit files in VSCode which have opencv/gocv imported, linting stops working because of errors in imgproc.cpp:
`error: expected '(' for function-style cast or type construction`

OS Info:
```
System Version:	macOS 10.13.4 (17E202)
Kernel Version:	Darwin 17.5.0
```

deadprogram, 2018-05-05:
Hi @jyzadeh you might want to try the `dev` branch of GoCV to resolve the syntax on macOS linting.

I have a few time needed to restart the program (not the drone) to get the video to start, might be some timing issue not yet fully understood.

jyzadeh, 2018-05-05:
Is there some installation difference between the master and dev branch? How would I switch to using dev branch?

deadprogram, 2018-05-05:
You now appear to be missing the correct path for linking OpenCV to GoCV. What change did you make, if you did not checkout the dev branch?

The easiest way to switch to dev branch for GoCV is probably:

```
cd $GOPATH/src/gocv/x/gocv
git checkout dev
git fetch
```

jyzadeh, 2018-05-05:
I fixed the three 
`expected '(' for function-style cast or type construction`
errors that I had by parenthesizing the types. 

I undid the changes and swapped to the dev branch, and have same issue as above.

deadprogram, 2018-05-05:
Do you happen to have more than 1 version of OpenCV installed via homebrew? Some people have seen this sort of thing before https://github.com/hybridgroup/gocv/issues/129

jyzadeh, 2018-05-05:
I've only got the latest version installed.
```
$ brew info opencv
opencv: stable 3.4.1 (bottled)
Open source computer vision library
https://opencv.org/
/usr/local/Cellar/opencv/3.4.1_5 (551 files, 97.8MB) *
  Poured from bottle on 2018-05-04 at 15:26:31
```

deadprogram, 2018-05-05:
You do realize you have to run the `env.sh` script once per session, right?

jyzadeh, 2018-05-05:
Yep. I’ve been running it every time I open terminal. Program and linter behavior remain consistent.

deadprogram, 2018-05-05:
I do not have that version of macOS installed, and do not have that issue. Other people have used GoCV successfully on macOS perhaps one of them can chime in.

jyzadeh, 2018-05-05:
What version are you on?

I’ll try and install Linux with parallels and see if I can get anything working there.

deadprogram, 2018-05-05:
I personally use Ubuntu 16.04

jyzadeh, 2018-05-06:
I've swapped over to an Ubuntu 16.04 environment. 
Everything compiles and runs properly now, and all go extensions for VSCode work. 

The only issue is when running any files that use the video feed (mplayer, ffmpeg, opencv), no window opens up to show the video. 

This is probably an issue with my Ubuntu rather than Go. 
Even with just the mplayer example, the code properly runs and connects to the drone, but no mplayer window opens. 
Do you have any ideas on whats going on?

jyzadeh, 2018-05-06:
```
$ netstat -l
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
...              
udp6       0      0 [::]:6038               [::]:*
```

Looks like the data is properly coming through. The only issue is the literal mplayer/opencv player window doesnt open. Could it be a permission or firewall issue?

jyzadeh, 2018-05-06:
oh man. I should have realized this sooner. 
The 
`drone.On(tello.VideoFrameEvent, func(data interface{}) {`
block is never executing. What would cause that?

jyzadeh, 2018-05-06:
Tracing the call hierarchy, it looks like the code is getting stuck in
```
func (fd *FD) ReadFrom(p []byte) (int, syscall.Sockaddr, error) {
	if err := fd.readLock(); err != nil {
		return 0, nil, err
	}
	defer fd.readUnlock()
	if err := fd.pd.prepareRead(fd.isFile); err != nil {
		return 0, nil, err
	}
	for {
		n, sa, err := syscall.Recvfrom(fd.Sysfd, p, 0)
		if err != nil {
			n = 0
			if err == syscall.EAGAIN && fd.pd.pollable() {
				if err = fd.pd.waitRead(fd.isFile); err == nil {
					continue
				}
			}
		}
		err = fd.eofError(n, err)
		return n, sa, err
	}
}
``` 
in fd_unix.go

It gets to
`
if err = fd.pd.waitRead(fd.isFile); err == nil {
`
and never gets a return from that waitRead function.

jyzadeh, 2018-05-06:
I receive data for FlightDataEvent, just not for the VideoFrameEvent. 

How can I check why no data is coming through to 6038?

deadprogram, 2018-07-10:
Hi @jyzadeh have you tried lately with the latest versions of Gobot, GoCV, and the Tello firmware?

closed 2019-07-08 by jakeyzadeh