hybridgroup/gobot #530: How to assign requested address for tello driver example

zlederman, 2018-05-14:
Hello,
I ran the tello driver_test.go program and it connects to wifi and when I originally ran it it threw 
`imported and not used: "fmt"`
I fixed this by adding a blank identifier infront of "fmt"
Now whenever I run the program it gives me 
`2018/05/14 10:45:53 Initializing connections...
2018/05/14 10:45:53 Initializing devices...
2018/05/14 10:45:53 Initializing device Tello-7F66FA413B622CC0 ...
2018/05/14 10:45:53 Robot tello initialized.
2018/05/14 10:45:53 Starting Robot tello ...
2018/05/14 10:45:53 Starting connections...
2018/05/14 10:45:53 Starting devices...
2018/05/14 10:45:53 Starting device Tello-7F66FA413B622CC0...
2018/05/14 10:45:53 Starting work...
`
then I guess it connects to the drone  but after a few minutes of nothing happening on the drone I get
`stick command error: write udp 192.168.10.2:8888->192.168.10.1:8889: write: can't assign requested address` over and over and over again and then the wifi connection for the TELLO-XXXXXX drops
so what is happening here and what can I do to fix it?
Thank you,
Zach

deadprogram, 2018-05-14:
Hi @zlederman you normally run tests in Go by running:

```bash
go test ./platforms/dji/tello
```

For which you should obtain the result:

```bash
ok      _/home/ron/Development/gobot/gobot/platforms/dji/tello  0.001s
```

Files in Go named `*_test.go` are literally tests, not example programs. You probably want to look at the examples directory for example, the file https://github.com/hybridgroup/gobot/blob/master/examples/tello.go

Hope that helps!

zlederman, 2018-05-14:
THANK YOU SO MUCH <3

deadprogram, 2018-05-24:
I'm going to close this now. Please reopen if needed.

closed 2018-05-24 by deadprogram