hybridgroup/gobot #545: opencv_window sample doesn't work

tomo0310, 2018-05-19:
Hi.
I want to use opencv_window sample. But it doesn't work.
https://gobot.io/documentation/examples/opencv_window/

OS:OSX 10.12.6

`2018/05/19 17:32:54 Initializing devices...
2018/05/19 17:32:54 Initializing device Window ...
2018/05/19 17:32:54 Initializing device Camera ...
2018/05/19 17:32:54 Robot cameraBot initialized.
2018/05/19 17:32:54 Starting Robot cameraBot ...
2018/05/19 17:32:54 Starting connections...
2018/05/19 17:32:54 Starting devices...
2018/05/19 17:32:54 Starting device Window...
2018/05/19 17:32:54 Starting device Camera...
2018/05/19 17:32:55 Starting work...
2018-05-19 17:32:55.728 opencv_window[6016:520299] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'`

I saw that issue.
https://github.com/hybridgroup/gocv/issues/31

So, I Add this code to the main function.But It doesn't work too. 
`runtime.LockOSThread()`

Does anyone know how to solve the problem?

emf42, 2018-05-20:
For Mac OS, all opencv code must be run from the main thread, so using the goroutines provided by gobot are out of the question. 
Use this instead https://github.com/hybridgroup/gocv/blob/master/cmd/capwindow/main.go

tomo0310, 2018-05-23:
Thank you for your answer. I finally want to run tello_opencv.go. It doesn't work because of the same reason. I'll use my old windows PC until opencv for Mac OS supports multi thread.

gen2thomas, 2023-07-07:
Since this is now closed in [Allure](AllureOfTheStars/Allure/issues/79) - is it also fixed for gobot?