hybridgroup/gobot #547: Joystick: Dualshock 4 Mapping Off?

SMerrony, 2018-05-19:
Just got a Sony Dualshock 4 and the mappings in platforms/joystick/joystick_dualshock4.go seem to be way off.

Checking with a couple of tools seems to confirm this.

Happy for me to fix this?

SMerrony, 2018-05-20:
It's trivial to correct the existing code - I just wanted to make sure that others are also seeing wrong mappings with the current version. Eg, buttons mixed up, RH stick not mapped etc.

deadprogram, 2018-05-20:
I've been using my DS3 controllers mostly, so it is very likely that whatever you are seeing is whatever everyone is. In other words, PR appreciated please!

deadprogram, 2018-05-31:
Version 1.11.0 has just been released that includes these fixes. Now closing, thank you everyone!

closed 2018-05-31 by deadprogram

mulysatest, 2018-07-27:
this code using this joystick lib, but for some reason it doesn't map the dualshock4 joystick properly on OSX any idea? 
https://www.drone-engineer.com/articles/76