hybridgroup/gobot #558: Recover support

Coolerfall, 2018-05-30:
Hi guys, 
Thanks for sharing this awesome project. I got `panic` for some reason whatever, and the program exited. I checked the source code, there's no ·recover· for `panic` sutiation. So any plan to add `recover` support to keep gobot alive?

jadr2ddude, 2018-05-30:
Can you please share where the panic came from? In go, panic/recover is usually bad.

trevrosen, 2018-05-30:
Are you wanting to see it as like a middleware in `api`?

Coolerfall, 2018-05-30:
@jadr2ddude, some foo bar panic,  maybe we didn't get it when testing. There always come some unknown bugs and maybe these bugs will crash.
@trevrosen, yes. Maybe we should keep them alive for further updates.

jadr2ddude, 2018-05-30:
While I do think this is possible to implement, this definitely should definitely not be the default behavior. Panic is considered bad practice, with the exception of 2 conditions:
1. Conditions indicating that there is an unrecoverable error and it is impossible to continue
2. Programmer error (e.g. null deference/index out of bounds)

In neither of these conditions is it safe to continue. If you hit a bug and decide to continue, there is an indeterminate state which could/would result in a cascade of more bugs.

trevrosen, 2018-05-30:
(cc @deadprogram since this is potentially a fairly fundamental discussion)

We could think through some middleware-style patterns that let us intercept and recover from a panic in situations where that makes sense. Would need to be configurable, I'd be in favor of offering such an option.

It can be argued that the robot work construct in Gobot is very similar to an `http.HandlerFunc` in the sense that it's a chunk of code whose execution really shouldn't negatively affect the supervisory entity. It's very common practice in Go (and many other langs) to handle exceptions in such a way that a server doesn't terminate but merely logs the issue. See [Negroni](https://github.com/urfave/negroni/blob/master/recovery.go) or [grpc-ecosystem's recovery interceptor](https://github.com/grpc-ecosystem/go-grpc-middleware/tree/master/recovery) for examples of recovery concepts in a server context.

trevrosen, 2018-06-02:
Thinking out loud here:

- We could start by adding some basic recover-and-log to the [Robot work goroutine](https://github.com/hybridgroup/gobot/blob/master/robot.go#L169-L172). This would enable the user to do something with their failure panic traces before the program exits and could be very useful for anyone working with a collection of Robots, especially in some scenario where they're set up to do remote work and an operator wants to collect a mass of diagnostic info.

- Past that, we could extend with some re-try functionality. This would enable the supervisor function (`Robot.Start()`) to re-start the `Robot.Work` function a configurable number of times, logging each crash via the previous functionality. This would be useful in scenarios where failure conditions inside a `Robot.Work` function are rare or acceptable.

deadprogram, 2018-06-05:
I like where this is going. We can certainly also allow the developer to disable any auto-recovery in the case they are using "metal Gobot", but I suspect that most devs want it to "just work".

trevrosen, 2018-06-05:
@deadprogram "metal Gobot" is basically just directly accessing hardware using provided drivers and connectors, right? If so, yeah I'd not even want to touch that line of use cases here. I'm specifically thinking about making the `Robot` orchestration concept more robust in the face of specific failure modes. I'll have something whipped up for you to react to this weekend. 😃

trevrosen, 2018-06-10:
@deadprogram here's a [first stab](https://github.com/hybridgroup/gobot/compare/dev...trevrosen:feature/558/recover-support?expand=1) at what recovery support might look like. Things to note:

* **this is still a work in-progress**
* A general API improvement here is the addition of support for [functional options](https://dave.cheney.net/2014/10/17/functional-options-for-friendly-apis) to `Robot`. Due to the nature of how `NewRobot` takes arguments, you add these and then call them with `AddConfigFunc()` and `RunConfigFuncs()` after initializing a new `Robot`.
* `Robot` now also has a mutex-guarded `lastError` field that can be set with `SetLastError(error)`. This enables a panicking `Work` function to note a failure before exit. This isn't super useful right now but **1)** it gives me something relatively easy/meaningful to test and **2)** it would be very useful if/when we take the next step and implement retries.
* I added a `sync.WaitGroup` on `Robot` to make testing of concurrent code possible/easier. As a bonus, this makes it simpler to synchronize other concurrent Go code along w/ a Robot's work.
* There's a new `_gobot_test` directory that implements a `gobot_test` package. There is a single test in there now showing how a user would work with the new config func/panic handler stuff. But this also represents the beginning of some over-all refactoring I'd like to do on some of our tests:
  * Remove usage of the `gobottest` package in favor of the more widely known and capable [Testify](https://github.com/stretchr/testify) lib
  * Leverage the `TestMain` function for global init/teardown needs
  * Test the `gobot` package's primary functionality from outside the package itself, helping to enforce some API boundaries.

### Further Discussion/Questions

* Should we leave `Stop()` to be called within the panic handler function itself as I've done in the test **or** should we take the more restrictive (but possibly more responsible) approach of automatically calling `Stop()` once the panic handler has executed? **This is the key question of this new functionality, IMO.**
* How do we feel about the new WaitGroup support? I **think** the code as written is fine to keep this from breaking existing instances of `Robots` people have, but we should double-check before I finalize this into a PR.
* Anything I've missed in this overall approach?