hybridgroup/gobot #560: Unable to compile raspi sample from docs

gauravagarwalr, 2018-06-05:
Hi, I am not sure what exactly is causing this:

For me I am trying to compile the sample code @ https://gobot.io/documentation/platforms/raspi/

It doesn't compile and gives the following error:

    vendor/gobot.io/x/gobot/drivers/spi/spi.go:82:21: cannot use maxSpeed (type int64) as type physic.Frequency in argument to p.Connect

The code for the above is located @ https://github.com/gauravagarwalr/gobot-ble-example/blob/master/main.go

Please help!

deadprogram, 2018-06-06:
This is due to a breaking change in the periph package used for Gobot's SPI support. You can solve this by installing the 2.3 version of periph:

```shell
cd $GOPATH/src/periph.io/x/periph
git fetch
git checkout v2.3.0
```

Now you can change back to your Gobot directory and should be able to build correctly. There is also a PR not yet merged with the same change but to the dep file #561

gauravagarwalr, 2018-06-06:
To anyone using `dep` & coming to this issue, add the following to your `Gopkg.toml`:

```
[[override]]
  name = "periph.io/x/periph"
  version = "2.3.0"
```

It fixes the issue for now.

gen2thomas, 2022-12-11:
@deadprogram it seams this is outdated, because we use v3.x of "periph.io/x/periph" now. So I'm going to close this issue. Please reopen it, if I'm wrong.

closed 2022-12-11 by gen2thomas