hybridgroup/gobot #562: Best way to detect if a firmata connection is established?

sdistefano, 2018-06-05:
I'm having problems detecting if my firmata connection is established or not. In either case, robot.Running() returns true and I can run log commands from a gobot.Every.

What's the recommended way to constantly check that the connection is established and attempt to restart it otherwise?

deadprogram, 2018-06-06:
The Robot's `Start()` currently ensures that all the `Adaptors` are correctly started. However it does NOT do anything about restarts/failover etc.

There is a conversation started in #558 about this very topic. I think this issue can be closed in favor of tracking that one.

sdistefano, 2018-06-06:
Thank you! Is there no known workaround at the moment? I have tried making a digital read as a form of PING, but even this is returning normal values even when I unplug the card after connection