hybridgroup/gobot #565: Stopping and Starting Robot with Raspberry Pi Adapter and LED Driver - LED does not toggle on restart

acarlonvs, 2018-06-10:
**Summary**
Starting, Stopping and Starting robot with Raspi Adapter and LED driver does not work on the second Start(). The LED does not toggle and Toggle returns "pin has not been exported".

**Cause**
Given the sample code from [https://gobot.io/documentation/platforms/raspi/](url): 

```
func main() {
        r := raspi.NewAdaptor()
        led := gpio.NewLedDriver(r, "7")
        work := func() {
                gobot.Every(1*time.Second, func() {
                        led.Toggle()
                })
        }
        robot := gobot.NewRobot("blinkBot",
                []gobot.Connection{r},
                []gobot.Device{led},
                work,
        )
        robot.Start()
}
```
If the robot is stopped and then restarted, the LED does not blink. The reason is in raspi_adapter.go. When the robot is stopped raspi_adapter.go `Finalize()` will `Unexport` the pins, but the pins are retained in the map. So, when the robot is again started and `DigitalWrite() -> DigitalPin() -> getExportedDigitalPin()` is called, we get to the following code: 

```
        if r.digitalPins[translatedPin] == nil {
		r.digitalPins[translatedPin] = sysfs.NewDigitalPin(translatedPin)
		if err = r.digitalPins[translatedPin].Export(); err != nil {
			return
		}
	}
```
Note, the pin is still in the map, but is now unexported. This results in the "pin has not been exported" error.

**Fix** 

First remove the map initialization from `NewAdaptor()`
```
 func NewAdaptor() *Adaptor {
        r := &Adaptor{
-               mutex:       &sync.Mutex{},
-               name:        gobot.DefaultName("RaspberryPi"),
-               digitalPins: make(map[int]*sysfs.DigitalPin),
-               pwmPins:     make(map[int]*PWMPin),
+               mutex: &sync.Mutex{},
+               name:  gobot.DefaultName("RaspberryPi"),
 }
```
Second, add to `Connect()`: 
```
 func (r *Adaptor) Connect() (err error) {
+       r.digitalPins = make(map[int]*sysfs.DigitalPin)
+       r.pwmPins = make(map[int]*PWMPin)
        return
 }
```
This will ensure that a new map is used each time the Robot is started. Not sure if there are any unintended effects of doing this, but at least it highlights what the problem is. 
I can try to do a pull request on dev with this change if that is helpful.

deadprogram, 2018-06-10:
Hi @acarlonvs I think this sounds like a very good idea. You are correct, that some tests will need to be changed to accommodate this. It also seems to me that this probably needs to be consistent across all Adaptors that support the various GPIO interfaces.

Anyone else want to weigh in on this?

acarlonvs, 2018-06-10:
@deadprogram, judging by the comments it seems that at one point Connect() was responsible for creating the digital and pwm pin adaptor maps:

```
// Connect starts connection with board and creates
// digitalPins and pwmPins adaptor maps
func (r *Adaptor) Connect() (err error) {
	return
}
```

acarlonvs, 2018-06-11:
@deadprogram, also PWM does not work quite right. Currently, pwm_pin.go does this: 
```
func (p *PWMPin) Unexport() error {
	return p.piBlaster(fmt.Sprintf("release %v\n", p.pin))
}
```
`release` will actually remove this pin from the known pins in pi blaster - see `release_pin` in [https://github.com/sarfata/pi-blaster/blob/master/pi-blaster.c](url)
This means when the Robot is started, the pin can no longer be used. Error on pi-blaster is "GPIO <gpio pin> is not enabled for pi-blaster."
 
Ideally it would be possible to call `release_pin2gpio`, but since this is not available, the next best is to set the PWM to 0: 
```
func (p *PWMPin) Unexport() error {
	return p.piBlaster(fmt.Sprintf("%v=%v\n", p.pin, 0))
}
```
A workaround would have been to set the duty cycle to 0, but pwm_pin.go does not allow it. there is the block: 
```
func (p *PWMPin) SetDutyCycle(duty uint32) (err error) {
        ...
	// never go below minimum allowed duty for pi blaster
        if val < 0.05 {
		val = 0.05
	}
```
This seems incorrect, 0 should be allowed to turn off PWM without Stopping robot.

gen2thomas, 2022-11-27:
Fixed for digital pin with #887 . This was tested with sysfs and character device driver. The fix is done in the new "DigitalPinAdaptor", which is included now by "composition" in the raspi adaptor.

Following platforms are also affected, most likely:
* beaglebone
* chip
* dragonboard (no pwm)
* edison
* joule
* jetson
* tinkerboard
* up2

gen2thomas, 2022-12-05:
The release part of pi-blaster was already fixed. The PWM pin cache problem is now fixed for all affected platforms with #890 .

gen2thomas, 2023-05-15:
part of release v2.0.0

closed 2023-05-15 by gen2thomas