hybridgroup/gobot #566: Unable to fetch >1.9.0 via vgo

pepijnschildkamp, 2018-06-12:
Hello,

First off, I know vgo is not production use ready. But i'm doing an hobby project on the Tello drone using this library for the SDK (which is great, thanks!). When fetching dependencies via `vgo get` it will install v1.9.0 as latest. Though the Tello SDK is added in the v1.10.0 version.

When fetching gobot via `go get` it will install the latest (v1.11.0).

Could the tags used in git (v1.9.0 vs 1.11.0 (notice the v)) cause this issue?

Regards

deadprogram, 2018-06-12:
Hello @pepijnschildkamp the short version is that I have not used `vgo` yet at all, so I do not know the answer from experience.

However according to the docs, you should be able to do this:

```shell
vgo get gobot.io/x/gobot@1.11.0
```

Please let us know how it works out!

pepijnschildkamp, 2018-06-12:
That doesn't work. See output below

```vgo: finding gobot.io/x/gobot v1.11.0
vgo: gobot.io/x/gobot v1.11.0: unknown revision "v1.11.0"
vgo: finding gobot.io/x/gobot v1.11.0
vgo: gobot.io/x/gobot v1.11.0: unknown revision "v1.11.0"
vgo get: unknown revision "v1.11.0"```

The unknown revision give me the idea vgo is looking for a tag called v.1.11.0. Which explains why it can find the v1.9.0 but not the tag 1.11.0.

You could consider adding an go.mod file to support vgo in the future.

pepijnschildkamp, 2018-06-12:
I will address this issue to the vgo community. I will use Dep for now for versioning. Thanks for your response.

closed 2018-06-12 by pepijnschildkamp