hybridgroup/gobot #570: how do I resubscribe after reconnect?

neewbee, 2018-06-22:


neewbee, 2018-06-22:
```golang
// +build example
//
// Do not build by default.

package main

import (
	"fmt"
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/platforms/mqtt"
)

func main() {
	mqttAdaptor := mqtt.NewAdaptor("tcp://localhost:1883", "pinger")

	work := func() {
		mqttAdaptor.On("hello", func(msg mqtt.Message) {
			fmt.Println("hello")
		})
		mqttAdaptor.On("hola", func(msg mqtt.Message) {
			fmt.Println("hola")
		})
		data := []byte("o")
		gobot.Every(1*time.Second, func() {
			mqttAdaptor.Publish("hello", data)
		})
		gobot.Every(5*time.Second, func() {
			mqttAdaptor.Publish("hola", data)
		})
	}

	robot := gobot.NewRobot("mqttBot",
		[]gobot.Connection{mqttAdaptor},
		work,
	)

	robot.Start()
}
```
several seconds after stopping mqtt server, reconnect will happen. But Work no longer works

deadprogram, 2018-06-22:
@trevrosen this is one of the things we want to be able to solve with your latest work in progress.

trevrosen, 2018-06-22:
@deadprogram indeed! Speaking of which you want me to proceed to PR w/ the implementation basically as it stands? Should we talk through some implications for existing installs etc?

grtfou, 2018-12-19:
I use two settings and it looks fine to my case. (gobot v1.12.0)
Maybe it's a solution if we're hard to monitor connection status.
```go
mqttAdaptor.SetAutoReconnect(true)
mqttAdaptor.SetCleanSession(false)
```

trevrosen, 2018-12-19:
@neewbee sorry for my silence on this -- it fell through the cracks. There's a new construct in the `dev` branch called RobotWork which gives you some external control of the workloads a robot is using. Please have a look at that and let me know (via this thread) whether it gives you the semantics you need to help solve your issue.

trevrosen, 2019-01-03:
Hi @neewbee I believe that [this PR](https://github.com/hybridgroup/gobot/pull/650) may also help you. I'm going to close this issue for now.

closed 2019-01-03 by trevrosen