hybridgroup/gobot #575: MCP2515 CAN Controller Driver and CAN Interface

ssnover95, 2018-06-25:
Hello,
I'm an embedded engineer and I stumbled on Gobot while searching around for a good framework for developing hardware in order to learn Go. I'd like to contribute by developing a driver for the MCP2515 which is a useful IC which lets a device communicate with a CAN network over SPI.

However, I'm aware that Go has some different styles and idioms. My thought was that this driver should also be paired with an implementation of some kind of CAN interface such that further implementation built on top could be agnostic with respect to whether CAN access is courtesy of this IC in the case of say a Raspberry Pi or internal CAN controller hardware in the case of the BeagleBone Black.

Are there any plans to implement a CAN interface in Gobot and if not any suggestions on designing this interface in a way idiomatic to Go?

jadr2ddude, 2018-06-26:
For builtin CAN devices handled by OS, x/sys/unix should be used for this interface:

https://godoc.org/golang.org/x/sys/unix#SockaddrCAN

ssnover95, 2018-06-28:
Ah good to know there's already existing utility for this. I've used SocketCAN before so I know it also works on the Raspberry Pi with this CAN controller IC.

closed 2018-06-28 by ssnover95