hybridgroup/gobot #579: GetSpiConnection function can't use maxSpeed (int64)

gilgameshskytrooper, 2018-07-05:
~/go/src/gobot.io/x/gobot/drivers/spi/spi.go|82 col 21 error| cannot use maxSpeed (type int64) as type physic.Frequency in argument to p.Connect

When I compile, I get this error.

gilgameshskytrooper, 2018-07-05:
drivers/spi/spi_test.go|11 col 42 error| cannot use TestSpiConnection literal (type *TestSpiConnection) as type spi.PortCloser in argument to NewConnection: *TestSpiConnection does not implement spi.PortCloser (wrong type for Connect method) have Connect(int64, spi.Mode, int) (spi.Conn, error) want Connect(physic.Frequency, spi.Mode, int) (spi.Conn, error)
drivers/spi/spi_test.go|11 col 60 error| cannot use TestSpiDevice literal (type *TestSpiDevice) as type spi.Conn in argument to NewConnection: *TestSpiDevice does not implement spi.Conn (missing String method)

gilgameshskytrooper, 2018-07-05:
Okay thanks.

gilgameshskytrooper, 2018-07-08:
Just for clarification, I am using the packages in gobot.io when I ran into these issues rather than github.

ssnover95, 2018-07-08:
I'm also running into this issue while importing the gobot.io/x/gobot package. I am not even using the SPI driver.

gilgameshskytrooper, 2018-07-09:
There is no dev branch in the git repo on gobot.io

deadprogram, 2018-07-09:
@gilgameshskytrooper that is a custom or "vanity" domain in Go. In order to check out the `dev` branch:

```
cd $GOPATH/src/gobot.io/x/gobot
git checkout dev
```

Hope that helps!