hybridgroup/gobot #587: Debugging message "Axis: 1" with PS4 joystick flood the screen

mulysatest, 2018-07-27:
Once I connected PS4 to computer and execute ./scanner  binary.
Seems like the left stick always keep sending signal and flood the screen like this
Any idea to fix this? 

```
[1188 ms] Axis: 1	value:-1414
[1222 ms] Axis: 1	value:-1671
[1223 ms] Axis: 1	value:-1414
[1239 ms] Axis: 1	value:-1671
[1256 ms] Axis: 1	value:-1414
[1276 ms] Axis: 1	value:-1671
[1314 ms] Axis: 1	value:-1414
[1314 ms] Axis: 1	value:-1157
[1331 ms] Axis: 1	value:-1414
[1367 ms] Axis: 1	value:-1928
[1367 ms] Axis: 1	value:-1671
[1384 ms] Axis: 1	value:-1928
[1401 ms] Axis: 1	value:-1414
[1435 ms] Axis: 1	value:-1671
[1435 ms] Axis: 1	value:-1414
[1451 ms] Axis: 1	value:-1671
[1470 ms] Axis: 1	value:-1414
[1487 ms] Axis: 1	value:-1928
[1488 ms] Axis: 1	value:-1414
[1540 ms] Axis: 1	value:-1928
[1560 ms] Axis: 1	value:-1414
[1580 ms] Axis: 1	value:-1671
[1597 ms] Axis: 1	value:-1414
```

deadprogram, 2018-08-02:
Hello @mulysatest you might want to verify that you are using the latest release of SDL. Which OS and version are you using?

deadprogram, 2018-08-16:
If you are using Linux, you should update your kernel to 4.14+, as I just discovered on a fresh Raspbian install. I ran into the same behavior with SDL2 2.0.8 and the latest Gobot release.

Hope that helps!

closed 2019-01-03 by trevrosen