hybridgroup/gobot #590: digispark i2c

vendelin8, 2018-08-09:
Hi,

I'd like to use PCA9685Driver with Digispark, through i2c. I checked the examples, code, documentations, but I can't figure it out, how to connect them.
I tried with this:

	digisparkAdaptor := digispark.NewAdaptor()
	pca9685 := i2c.NewPCA9685Driver(digisparkAdaptor.connector, i2c.WithAddress(0x27))

but it gives me:

    digisparkAdaptor.connector undefined (type *digispark.Adaptor has no field or method connector)

Please help!

vendelin8, 2018-08-11:
I started to write the missing parts, but it doesn't seem to work.
https://github.com/hybridgroup/gobot/compare/master...vendelin8:master

I'm using this code to get it work:

	digisparkAdaptor := digispark.NewAdaptor()
	pca9685 := i2c.NewPCA9685Driver(digisparkAdaptor, i2c.WithAddress(0x27))

	work := func() {
		val := 0
		gobot.Every(50 * time.Millisecond, func() {
			val += 50
			if val > 4096 {
				val = 0
			}
			pca9685.SetPWM(0, 0, val)
			fmt.Println(val)
		})
	}

	robot := gobot.NewRobot("pwmBot",
		[]gobot.Connection{digisparkAdaptor},
		[]gobot.Device{pca9685},
		work,
	)

	defer func() {
		pca9685.SetPWM(channel, 0, 0)
	}()
	robot.Start()

When I tried to add

    pca9685.SetPWMFreq(60)

It broke with:

    panic: runtime error: invalid memory address or nil pointer dereference
    [signal SIGSEGV: segmentation violation code=0x1 addr=0x40 pc=0x4af58c]

    goroutine 1 [running]:
    gobot.io/x/gobot/drivers/i2c.(*PCA9685Driver).SetPWMFreq(0xc4200620c0, 0xc442700000, 0xc420057ef8, 0x1)

I'm quite new to gobot, cgo and low level programming, that's what I could figure out by myself. If you have some time, please look at it. Thanks!

vendelin8, 2018-08-17:
I found this example, that helped me with the issue:
https://github.com/littlewire/Little-Wire/blob/master/software/examples/i2c_blinkM.c

So I improved the code. I added a Test functionality that is able to check available addresses, with a code like:

    digisparkAdaptor.TestConnection(0, 127, func(a int) {
        fmt.Println(a)
    })

It gives me 0 and 64 (=0x40), so I changed the calling code to this

    pca9685 := i2c.NewPCA9685Driver(digisparkAdaptor, i2c.WithAddress(0x40))

But it still doesn't work, not sure why. Still waiting for any help, please.

deadprogram, 2018-08-22:
Hello @vendelin8 in order for an Adaptor to support i2c it needs to implement both the `I2cOperations` interface https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/i2c.go#L29 as well as the `Connector` interface https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/i2c.go#L48

You can verify this is complete if you add the following lines here https://github.com/hybridgroup/gobot/blob/master/platforms/digispark/digispark_adaptor_test.go#L18

```
var _ i2c.I2cOperations = (*Adaptor)(nil)
var _ i2c.Connector = (*Adaptor)(nil)
```

vendelin8, 2018-08-22:
Ok, I added

    var _ i2c.Connector = (*Adaptor)(nil)

to the test file, and wrote some new tests as well. I think that the other line shouldn't be necessary. Because:
1. not the Adaptor itself, but the connection returned by `GetConnection` should implement `i2c.I2cOperations`.
2. if a connection returned by `GetConnection` implements `i2c.Connection`, it will implement `i2c.I2cOperations` also.
3. I grepped the whole project for `I2cOperations`, and I found it only in https://github.com/hybridgroup/gobot/blob/master/drivers/i2c/i2c.go

So my current branch looks like this:
https://github.com/hybridgroup/gobot/compare/master...vendelin8:master

It compiles, tests are passing, but it doesn't seem that my PCA9685 sends appropriate PWM signals. Possible errors are:
1. I made something wrong in the code.
2. I failed to connect hardware parts properly. Not very likely, but I'm really new to this field.

deadprogram, 2018-08-22:
Take a look at https://github.com/littlewire/littlewire.rb/blob/master/lib/i2c.rb for a working client lib that uses the i2c interface of littlewire.

vendelin8, 2018-08-23:
Nice, thanks!
https://github.com/hybridgroup/gobot/pull/598

deadprogram, 2018-08-24:
Hi @vendelin8 thank you very much for the PR. I made a few small tweaks, but it worked for me with 2 different i2c devices:

- BlinkM
- MPL115A2

I flashed my DIgispark with Littlewire firmware using Gort.

Then I connected my Digispark via P0 (SDA) P2 (SCL) 5V and GND pins to the i2c devices.

The 2 new examples that I wrote both worked as expected.

What I would suggest is to make sure that you have flashed your Digispark correctly, and also make sure that your device has 10K pullup resistors on both SDA and SCL if needed for your device.

I have now merged the PR thank you again for working on this!

I will close this issue after the next release of Gobot.

vendelin8, 2018-08-25:
Ok, finally I was able to make it work. Thanks for the assistance, and for this awesome tool! :)