hybridgroup/gobot #593: Allow SetPeroid() in raspi/pwm_pin.go

starryalley, 2018-08-14:
Summary:

I think to correctly use pi-blaster on raspberry pi, SetPeriod() in platforms/raspi/pwm_pin.go should be allowed to change the now-constant value of piBlasterPeriod. 

On my board I modified pi-blaster CYCLE_TIME_US to suit my application. Only when we allow SetPeroid() to change piBlasterPerioid value, can one correctly use SetDutyCycle() to write correct duty cycle to pi-blaster.

For example, my CYCLE_TIME_US is set to 20000us (which is 20000000ns, currently it's fixed to 10000000). To set 1.5ms PWM signal the correct command to write to `/dev/pi-blaster` is `18=0.075`.  (where 18 is the pin 12). However, the following code
```
pin, _ := r.PWMPin("12")
pin.SetDutyCycle(1.5 * 1000 * 1000)
```
will write `18=0.15`.

deadprogram, 2018-08-15:
Hello, @starryalley that seems like a very useful, and modest change. Perhaps we can just change `piBlasterPeriod` to a public member of the Adaptor ala `PiBlasterPeriod` so it can be set before the work routine starts.

I'd be happy to accept a PR with this update!

starryalley, 2018-08-15:
Thanks for the suggestion and also this great project. I just sent a PR https://github.com/hybridgroup/gobot/pull/594

Let me know what you think.

starryalley, 2018-08-27:
PR merged. Closing.

closed 2018-08-27 by starryalley