hybridgroup/gobot #596: Beginner questions regarding the Tello drone

luedigernet, 2018-08-18:
Hello community,
today I got my Tello drone and started to program it with gobot. And I'm wonder why the land command did not work. I've used several examples from the Blog side here but my drone will not react on the land command. Furthermore I wonder why I receive VideoFrame Events even if I did not send a StartVideo command.

kind regards

Reinhard

deadprogram, 2018-08-22:
Hello, @luedigernet 

You probably did not wait quite long enough before sending your Land() command after the takeoff. The example at https://github.com/hybridgroup/gobot/blob/master/examples/tello.go waits 5 seconds after take off.

The drone sends the video frames regardless of the StartVideo() command. StartVideo() actually tells the Tello to send the h264 headers that contain the info considered the start of the stream, which are needed to decode the video.

Hope that helps!

luedigernet, 2018-08-24:
Hi deadprogram,

thx for your comment.  The timing of the land command might be the issue so for now we can close the issue.

closed 2019-01-03 by trevrosen