hybridgroup/gobot #600: More than 2 Sphero SPRK+ devices doesn't work

nilskuhn, 2018-08-29:
Hi there,

first of all thanks a lot for the beautiful work. I have a lot of fun with it!

I've created a game server that receives ROLL / ROTATE / SET_RGB - Actions from mqtt topic's which than should be executed by different gobot controlled Sphero SPRK+ bots: https://github.com/iteratec/sphero-rallye-server

Everything works fine so far if I control up to two Spheros. But for more than 2 devices no Bluetooth LE connection gets established :(

I am running the server from my local laptop (Dell Latitude 7480) running kde neon / ubuntu. Can anybody say if I have to do something to establish more than two Bluetooth LE connections? Could it be a hardware limitation? I haven't found anything about it so far. Or is this a bug / problem in Gobot library? Anyone else experiences with Sphero swarms of more than two Sphero SPRK+ bots?

Thanks in advance, Nils

deadprogram, 2018-08-29:
Hello @nilskuhn you probably will need to update your kernel in order to use multiple BLE connections to at least 4.13+ or at least that was my experience.

nilskuhn, 2018-08-30:
```
$ uname -a
Linux iteratec-laptop 4.15.0-32-generic #35~16.04.1-Ubuntu SMP Fri Aug 10 21:54:34 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
By multiple you mean more than two? Because two are already working.

nilskuhn, 2018-09-03:
The problem was my laptop or the kernel version/os on it. 
On a raspberry pi version 3 it worked :)

closed 2018-09-03 by nilskuhn