hybridgroup/gobot #603: Incorrect data events when having multiple analog sensors in Raspberry Pi

mabu, 2018-08-31:
The step6 program of https://github.com/hybridgroup/gophercon-2018/tree/master/sensor/raspi/ had a couple issues:

- Rotary movement is ignored;
- When the light sensor value changes to X, two messages are produced: "Light: X" and "Dial: X".

If I change the order of devices on [line 75](https://github.com/hybridgroup/gophercon-2018/blob/c8c0466f1425505f0f7acfcb2928f1708b906087/sensor/raspi/step6/main.go#L75) from "dial, light" to "light, dial", then the other sensor works:

- Light change is ignored;
- When I turn the rotary and its value changes to X, two messages are produced: "Light: X" and "Dial: X".

So, it looks like only the sensor that is listed last in the device slice works, and its events are sent to the handlers of both sensors.

deadprogram, 2018-09-06:
This appears to be an issue with the GrovePi+ as I just brought up here: https://github.com/DexterInd/GrovePi/pull/423

deadprogram, 2019-05-22:
This code was release in v1.13.0 thank you! Now closing.

closed 2019-05-22 by deadprogram