hybridgroup/gobot #607: Bebop 2 video streaming via rtp requires undocumented setup

krancour, 2018-09-10:
NB: My drone is running the latest firmware.

It seems that without the following line, streaming via RTP does not work:

```
if err := bebop.VideoStreamMode(2); err != nil {
	log.Fatalf("Error setting stream mode to RTP: %s\n", err)
}
```

I stumbled upon this solution after lots and lots of googling, but video stream mode do not seem to be documented anywhere.

deadprogram, 2018-09-10:
Hello @krancour are you saying that it DOES work when you add that call? If so can you please submit a PR with that addition?

krancour, 2018-09-10:
After digging through lots of Parrot docs and discovering that stream mode has to do with latency and stream quality (and nothing to do with _method_ of streaming), I tried again and I can't reproduce this. Closing.

closed 2018-09-10 by krancour