hybridgroup/gobot #609: Bebop 2 doesn't always land when it should

krancour, 2018-09-10:
I'm using the client directly, e.g.:

```
// Connect
bebop := client.New()
if err := bebop.Connect(); err != nil {
	log.Fatalf("Connection error: %s\n", err)
}
log.Println("Connected. [Press ctrl+c to disconnect.]")
```

Later, I take off.

My program _ends_ with this:

```
// Land
if err := bebop.Land(); err != nil {
		log.Fatalf("Landing error: %s\n", err)
}
log.Println("We have touch down!")
```

While this program always exits without error, _sometimes_ after doing so, the drone continues flying. The first time this happened, I had to very quickly write a short, new "emergency landing" program that just issues the land command again.

But sometimes the program works as expected.

I am wondering if `bebop.Land()` might sometimes be returning prematurely, before the drone has ACK'ed the request to land.

deadprogram, 2018-09-10:
That is a very possible scenario. Please do report back with whatever you discover!

krancour, 2018-09-11:
@deadprogram I could be reading the bebop client code all wrong, but if I'm not, it looks a lot like the client doesn't receive any state, events, or even acks from the drone. Looking at similar libraries written in other languages, this seems like a conspicuous absence from this one... I'm not sure how to reasonably control a drone when I cannot query any of its state.

deadprogram, 2018-09-11:
The method `packetReceiver()` needs to be extended to parse the messages coming from the Bebop. Please see https://github.com/hybridgroup/gobot/blob/master/platforms/parrot/bebop/client/client.go#L526

Many of the values are already in https://github.com/hybridgroup/gobot/blob/master/platforms/parrot/bebop/client/constants.go and just need to get hooked up.

```go
if frame.Id == int(BD_NET_DC_EVENT_ID) {
    // parse the frame.Data header here
    project := uint(frame.Data[0])
    class := uint(frame.Data[1])
    commandId := int(frame.Data[2]) + int(frame.Data[3]) << 8
    
    // handle the now identified event...
}
```

For example, 

```go
if project == ARCOMMANDS_ID_PROJECT_ARDRONE3 && class == ARCOMMANDS_ID_ARDRONE3_CLASS_PILOTINGSTATE && commandId == ARCOMMANDS_ID_ARDRONE3_PILOTINGSTATE_CMD_FLYINGSTATECHANGED {
    // next 4 bytes contain the actual flying state as little-endian int32
}
```

https://github.com/hybridgroup/gobot/blob/master/platforms/parrot/bebop/client/constants.go

My Bebop drones are currently in storage so I'm sorry I cannot directly help much at the moment. Hopefully this can point you in the right direction.

krancour, 2018-09-11:
Thanks @deadprogram.

As I pour over docs, I'm finding this client somewhat incomplete and out-of-date (as you had acknowledged). It seems the ARNetwork/ARNetworkAL protocol is only partially implemented here-- for instance, there are no retries executed for frames that time out waiting for an ack. Logic for dropping duplicate and out-of-sequence frames in the manner described by the protocol is also missing.

I'm sure you can imagine how the above could be a critical factor in a request to land seemingly having gone unacknowledged.

I'm to actually try my hand at writing a complete ARNetwork/ARNetworkAL protocol library first then I'll retrofit existing commands to use that library. I'll also probably write a layer that consumes device --> client events (much as you suggested in your previous comment) and state information so that it will become possible to do things like "land and then wait for flying state to be 'landed'".

Last, if I nail all of the above (which is all pretty generic / protocol oriented) it may be possible for me pretty easily add client support for dozens of supported commands that aren't included today.

I think in the end, what I'm going to end up with is going to be indistinguishable from the client that exists now, so I'm going to start this effort as its own library and then maybe we can discuss later if / how to fold it back into gobot.

deadprogram, 2018-09-11:
You can certainly do that. However one thing that makes writing it within Gobot a LOT easier is having the whole event system built-in. Check out the Firmata adaptor/client for example of what I mean: https://github.com/hybridgroup/gobot/tree/master/platforms/firmata

In any case, keep us all informed on your progress.

krancour, 2018-09-11:
> However one thing that makes writing it within Gobot a LOT easier is having the whole event system built-in.

Totally agree. However, as it stands now, the bebop client, as you have noted, takes all events that the drone sends it over the ARNetwork protocol and drops them on the floor. Before I can think about propagating such events up to higher levels of abstraction (i.e. a robot), I need to implement some kind of event-handling within the client itself... that's all I was referring to.

deadprogram, 2018-09-12:
So turns out I have 1 of my Bebops in the closet here, not all off them in storage. I will try to take a look at some of this next week.

krancour, 2018-09-25:
@deadprogram I'm closing in on a more complete implementation of the arnetwork and arnetworkal protocols. These will actually wait for acks for outbound data that is intended to receive an ack.

When I am done, I'd be happy to donate those. It should be fairly easy to plug the existing client into that.

I am working on my own client as well because I need a client that receives/understand drone state and makes this available to the client's user. This may be of less interest to you, but I'd be happy to donate that as well if you wanted it.

closed 2021-09-30 by krancour