hybridgroup/gobot #610: Bebop 2 start/stop recording doesn't work as expected

krancour, 2018-09-10:
I'm running the latest firmware and I've encountered a few unexpected behaviors with the camera:

1. As soon as a flight begins, video is being captured. This is unexpected, as I would assume it's necessary to call `StartRecording()`.

1. In light of no. 1, I've attempted calling `StopRecording()` before take off-- it has no effect. Video is still captured.

1. Thinking it was possible (but not documented) that `StopRecording()` is only meant to be called after take off, I moved the call to immediately after takeoff. That, too, had no effect and video was captured for the entire duration of the flight.

Bottom line is it's not clear how, if at all, video capture can be suspended.

krancour, 2018-09-10:
@deadprogram it seems there's something in the API for setting whether video is automatically recorded from take off to a moment after landing:

https://github.com/Parrot-Developers/arsdk-xml/blob/master/xml/ardrone3.xml#L2220-L2234

The Bebop client in gobot doesn't seem to expose this. I think it would be easy enough for me to add, but I am trying to understand whatever code generation is in play for generating this client, as many of the functions have an appearance of having been generated.

Any chance of pointing me in the right direction?

deadprogram, 2018-09-10:
All of the Gobot code was written by hand, we do not have a code generator for this task. It certainly would be possible to create one...

krancour, 2018-09-10:
Well... if it's written by hand, that certainly makes it a little easier for me to wrap my head around how to expose this setting. :)

But just to confirm... in the following, example, `ARCOMMANDS_Generator_GenerateARDrone3PilotingTakeOff` in the comment is _not_ indicative of some generation?

```
func (b *Bebop) TakeOff() error {
	//
	//  ARCOMMANDS_Generator_GenerateARDrone3PilotingTakeOff
	//

	cmd := &bytes.Buffer{}

	cmd.WriteByte(ARCOMMANDS_ID_PROJECT_ARDRONE3)
	cmd.WriteByte(ARCOMMANDS_ID_ARDRONE3_CLASS_PILOTING)

	tmp := &bytes.Buffer{}
	binary.Write(tmp, binary.LittleEndian, uint16(ARCOMMANDS_ID_ARDRONE3_PILOTING_CMD_TAKEOFF))

	cmd.Write(tmp.Bytes())

	_, err := b.write(b.networkFrameGenerator(cmd, ARNETWORKAL_FRAME_TYPE_DATA, BD_NET_CD_NONACK_ID).Bytes())
	return err
}
```

deadprogram, 2018-09-10:
That comment was added by hand to provide some reference point back to the original place it came from, which was part of the old Parrot tools. It was never used to generate Go code.

krancour, 2018-09-10:
@deadprogram thanks for the explanation!

closed 2021-09-30 by krancour