hybridgroup/gobot #614: Does gobot currently support Raspberry Pi Zero/Raspberry Pi Zero Wireless?

gilgameshskytrooper, 2018-09-12:
I have tried the basic Pi example with the LED and I can't seem to get it to work.

The hardware layout I used was copied from [here](http://www.tom-bennett.co/article/blink-an-led-with-raspberry-pi-zero-and-python). The Python script they provided worked as advertised.

However, using the following code in Go, I cannot seem to get the LED to blink on my Raspberry Pi Zero with the following code:

```go
package main

import (
	"time"

	"gobot.io/x/gobot"
	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/raspi"
)

func main() {

	r := raspi.NewAdaptor()
	led := gpio.NewLedDriver(r, "17")

	work := func() {
		gobot.Every(1*time.Second, func() {
			led.Toggle()
		})
	}

	robot := gobot.NewRobot("blinkBot",
		[]gobot.Connection{r},
		[]gobot.Device{led},
		work,
	)

	robot.Start()

}
```
I also tried using pin "13" since technically, the board numbering is 13.

gilgameshskytrooper, 2018-09-12:
Also tried the idiomatic method.

```go
package main

import (
	"time"

	"gobot.io/x/gobot/drivers/gpio"
	"gobot.io/x/gobot/platforms/raspi"
)

func main() {

	r := raspi.NewAdaptor()

	led := gpio.NewLedDriver(r, "17")
	led.Start()

	for {
		led.Toggle()
		time.Sleep(1000 * time.Millisecond)
	}

}
```

(And I tried `13`)

lucavallin, 2018-09-16:
Not sure if the case, but remember to add these flags when compiling:

`GOOS=linux GOARCH=arm GOARM=6 go build`

I remember I had some issue because of the missing GOARM

gilgameshskytrooper, 2018-09-17:
I believe this was a hardware issue. Sorry to bother you guys!

closed 2018-09-17 by gilgameshskytrooper