hybridgroup/gobot #617: drivers/gpio: SA4000 warning in max7219_driver.go

Quasilyte, 2018-09-14:
Both gocritic and megacheck (staticcheck) find these code lines suspicious:

https://github.com/hybridgroup/gobot/blob/d7818a756eb3c2b275ddf9138e0e529dfb87f5dd/drivers/gpio/max7219_driver.go#L89
https://github.com/hybridgroup/gobot/blob/d7818a756eb3c2b275ddf9138e0e529dfb87f5dd/drivers/gpio/max7219_driver.go#L65

`&` is used with same RHS and LHS operands.

Should probably be simplified as `x&x`=>`x`, but as it looks somewhat confusing, I've opened an issue instead of sending a PR (maybe there is some IoT-related domain convention so `x&x` makes sense here).

deadprogram, 2018-09-16:
There is now code in the `dev` branch thanks to @conejoninja so I will leave this open until the next release. Thanks everyone!

deadprogram, 2019-05-22:
This code was release in v1.13.0 thank you! Now closing.

closed 2019-05-22 by deadprogram