hybridgroup/gobot #623: platforms/chip: quickstart sample/doc broken

ahmetb, 2018-09-24:
I'm following https://gobot.io/documentation/platforms/chip/ on macOS and seeing a compilation error in the sample program:

    undefined: gobot.On

## Repro steps

I ran `go get -d -u gobot.io/x/gobot/...` (worked)

Pasted the sample listed to main.go:

```
package main

import (
    "fmt"

    "gobot.io/x/gobot"
    "gobot.io/x/gobot/drivers/gpio"
    "gobot.io/x/gobot/platforms/chip"
)

func main() {
    chipAdaptor := chip.NewAdaptor()
    button := gpio.NewButtonDriver(chipAdaptor, "XIO-P0")

    work := func() {
        gobot.On(button.Event("push"), func(data interface{}) {
            fmt.Println("button pressed")
        })

        gobot.On(button.Event("release"), func(data interface{}) {
            fmt.Println("button released")
        })
    }

    robot := gobot.NewRobot("buttonBot",
        []gobot.Connection{chipAdaptor},
        []gobot.Device{button},
        work,
    )

    robot.Start()
}
```

then I ran `GOARM=7 GOARCH=arm GOOS=linux go build .` as instructed and this cmd gives error:

    ./main.go:16:9: undefined: gobot.On
    ./main.go:20:9: undefined: gobot.On

ahmetb, 2018-09-24:
similarly chip-pro instructions are broken, too https://gobot.io/documentation/platforms/chip-pro/

It seems like `gobot.On` isn't really a thing anymore. But I can't really find an `On` method for `platforms/chip` package to handle button press to refactor/fix the example code.

Skgamer38, 2020-03-25:
I have an arduino nano with a atmel atmega328p microcontroller and idk if it can emulate a HID, to act like a keyboard and mouse. Someone can help me?

gen2thomas, 2023-07-07:
Also there is a mismatch in dragonboard docu and example. I will fix this together.

gen2thomas, 2023-10-30:
available since v2.2.0

PR #977 (2023-07-07): dragonboard: fix example and documentation
## Solved issues and/or description of the change

Fix #623 

## Manual test

non

## Checklist

- [x] The PR's target branch is 'hybridgroup:dev'
- [x] New and existing unit tests pass locally with my changes (e.g. by run `make test`)
- [x] I have performed a self-review of my own code
closed 2023-10-30 by gen2thomas