hybridgroup/gobot #625: How would you "abstract" the readable devices?

lucavallin, 2018-09-27:
To overcome boilerplate I'd like to do something like this (or in the future use something like an abstract factory)

```
	var sensors map[string]gobot.Device
	sensors["sound"] = aio.NewGroveSoundSensorDriver(gp, "A0")
	sensors["temperature"] = aio.NewGroveTemperatureSensorDriver(gp, "A1")
	sensors["light"] = aio.NewGroveLightSensorDriver(gp, "A2")

	work := func() {
		gobot.Every(timeInterval, func() {
			for name, driver := range sensors {
				val, _ := driver.Read()
				json, _:= json.Marshal(messages.NewReading(name, val))
				mqttAdaptor.Publish("from", json)
			}
		})
	}
```

Obviously this won't compile because gobot.Device doesn't have a Read() method. Any idea?

lucavallin, 2018-10-05:
I see this is tagged as enhancement, but in the meantime, do you have any suggestions about how to wrap this up? Maybe by wrapping each sensor in some object which implements an interface, and then having a big list of sensors to loop through?

gen2thomas, 2023-07-07:
Hi @lucavallin , IMO this can be solved now by using "AnalogSensorDriver".

old: `var sensors map[string]gobot.Device`
new: `var sensors map[string]aio.AnalogSensorDriver`

Best regards Thomas

gen2thomas, 2023-07-07:
A more generic solution (interfaces in Go at client side):
```
type MyReader interface {
	Read() (float64, error)
}

func main() {
	gp := edison.NewAdaptor()

	sensors := make(map[string]MyReader)
	sensors["sound"] = aio.NewGroveSoundSensorDriver(gp, "A0")
	sensors["temperature"] = aio.NewGroveTemperatureSensorDriver(gp, "A1")
	sensors["light"] = aio.NewGroveLightSensorDriver(gp, "A2")
...
```

gen2thomas, 2023-11-10:
I'm going to close this issue, because this is more programming language related and do not affect gobot.

closed 2023-11-10 by gen2thomas