hybridgroup/gobot #639: lsm303DLHC Magnetometer output not updating

jobale, 2018-11-18:
I am currently trying to write a driver for the **lsm303DLHC** sensor. It's a part of [Adafruit 9DOF sensor](https://learn.adafruit.com/adafruit-9-dof-imu-breakout) composed of l3gd20h and [lsm303DLHC](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=2ahUKEwiEovu60d3eAhUszoUKHQbqDHkQFjAAegQIABAC&url=https://www.st.com/resource/en/datasheet/DM00027543.pdf&usg=AOvVaw14yHyIjS17hcLuL92Vm8S5).
I can successfully read l3gd20h sensor and the lsm303DLHC's accelerometer output but I have a problem with the magnetometer: I can read all 3 values, but they are not updated. I am stuck with the same outputs values.

I precise the sensor works perfectly with an arduino and Adafruit C++ library.

### Devices used

 - Raspberry Pi 2 with raspbian - i2c enabled - /dev/i2c-1 working
 - Adafruit 9DOF sensor composed of l3gd20h + lsm303DLHC

### Details
The magnetometer output is composed of 6 registers continuously updated.
The sensor automatically increment the register address when you read the output (default behavior). 

If I am not mistaken, here is how it goes:
 - Master write to 1st output register to initiate output reading
 - lsm303 device send back the 1st register output value
 - Master send **ACK** signal after reading the first output value
 - lsm303 receive the **ACK** signal, increment the register address and transmit the 2nd value
 - Master receive the 2nd output, etc...
 - lsm303 device send back the 6th and **last** output value and wait for a **NACK** signal this time.
 
 This is where the problem lies I think: If the lsm303 sensor receive a **NACK** signal it update the output values. If it receive an **ACK** signal, it doesn't update and just send the next output value (if any).
 So after sending the 6th and last register output, the sensor wait for a different signal before updating the outputs values. It wait for a **NACK** signal. Not an **ACK**.

 I based my opinion  [on this stackexchange post ](https://electronics.stackexchange.com/questions/101753/stm32f3discovery-lsm303dlhc-magnetic-field-data-not-updating) (read the last post)

Official [lsm303DLHC Datasheet](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=2ahUKEwiEovu60d3eAhUszoUKHQbqDHkQFjAAegQIABAC&url=https://www.st.com/resource/en/datasheet/DM00027543.pdf&usg=AOvVaw14yHyIjS17hcLuL92Vm8S5) doesn't mention any of this.

The lsm303DLHC works with an arduino. So I am wondering:
Does the raspberry pi 2 send a NACK signal automatically at the end of a buffer read ?
I looked at the [SMSBus and i2c documentation in the linux kernel](https://www.kernel.org/doc/Documentation/i2c/functionality) but I didn't found yet a definitive answer to my question.

I tried using [i2c_Device.go](https://github.com/hybridgroup/gobot/blob/master/sysfs/i2c_device.go) functions directly without success. I tried reading the value with *ReadByteData()* function. Nothing change. Same output every loop.

## My driver code to read the magnetometer output

const lsm303RegisterMagOutXLSB = 0x03

    // MagneticField returns the magnetic field value for the 3 axis
	// result's unit is in microTesla (mT)
	
	func (d *LSM303Driver) MagneticField() (x, z, y float32, err error) {
	if _, err = d.connection.Write([]byte{lsm303RegisterMagOutXLSB}); err != nil {
		return 0, 0, 0, err
	}

	measures := make([]byte, 6)
	if _, err = d.connection.Read(measures); err != nil {
		return 0, 0, 0, err
	}

	var rawXh, rawXl, rawZh, rawZl, rawYh, rawYl uint8
		
	buf := bytes.NewBuffer(measurements)

	binary.Read(buf, binary.BigEndian, &rawXh)
	binary.Read(buf, binary.BigEndian, &rawXl)
	binary.Read(buf, binary.BigEndian, &rawZh)
	binary.Read(buf, binary.BigEndian, &rawZl)
	binary.Read(buf, binary.BigEndian, &rawYh)
	binary.Read(buf, binary.BigEndian, &rawYl)
	...	
	x := ...
	y := ...
	z := ...
	return x, z, y, nil
}
This code is stripped of error check and unit conversion for readability.

Thanks for taking the time to read this. I tried to submit this problem the right way. I apologize if I miss or broke any rule doing it.

jobale, 2018-11-21:
I managed to get updated magnetic field values now. My code works from that point of view. My displaying of measures was the culprit. It wasn't precise enough to see the variations, hence the false impression of a never updated output. Nothing related to i2c protocol.

As for the raspberry pi 2 itself and i2c protocol I learn the following:

- It doesn't implement all SMBUS  functions on linux. To know which subset is available on your platform you can use this command on linux: `i2cdetect -F 1` (or i2cdetect -F X if you use bus X). Raspberry Pi2 lacks 2 functions: `SMBus Block Read` and `SMBus Block Process Call`.

- If you want know more about SMBus protocol you can read [this documentation on kernel.org.](https://www.kernel.org/doc/Documentation/i2c/smbus-protocol). It contains the breakdown of each SMBus function (ACK,NACK,STOP signals and more)

  

I still have to make sense of those outputs values. They are quite strange but this is another story ;-)
 This issue is without purpose now so I close it.

closed 2018-11-21 by jobale

gen2thomas, 2022-10-19:
Hi @jobale , thanks for pointing this out and try to provide a new driver. Sorry for the long delay. Meanwhile I have changed and fixed some i2c related behavior (currently available on "dev" branch). Possibly your driver will now work. It would be helpful if you can provide a branch or a Draft-PR to find some further problems.

jobale, 2022-10-19:
Hi @gen2thomas, Thanks for the attention. Just to inform hybridgroup/Gobot that I won't be able to help anymore on this issue: I don't write in Go anymore and I don't have lsm303DLHC chip anymore :-( . Sorry.

gen2thomas, 2022-10-19:
@jobale thanks for the fast reply. IMO the device request can be left open. I will try to order the module and implement the driver, based on your work. Thanks again!