hybridgroup/gobot #641: * The system cannot find the path specified.

recabyte008, 2018-11-27:
Install gobot and gort with firmata for Arduino on Windows 10.

go run the following sample code : 
package main

import (
        "time"
        "gobot.io/x/gobot"
        "gobot.io/x/gobot/drivers/gpio"
        "gobot.io/x/gobot/platforms/firmata"
)

func main() {
        firmataAdaptor := firmata.NewAdaptor("/dev/ttyACM0")
        led := gpio.NewLedDriver(firmataAdaptor, "13")

        work := func() {
                gobot.Every(1*time.Second, func() {
                        led.Toggle()
                })
        }
        robot := gobot.NewRobot("bot",
                []gobot.Connection{firmataAdaptor},
                []gobot.Device{led},
                work,
        )
        robot.Start()
}

Error Message : 

D:\go\projects>go run test1.go
2018/11/27 12:28:19 Initializing connections...
2018/11/27 12:28:19 Initializing connection Firmata-1FAE66071409C015 ...
2018/11/27 12:28:19 Initializing devices...
2018/11/27 12:28:19 Initializing device LED-76D7E1BEAB86A5DF ...
2018/11/27 12:28:19 Robot bot initialized.
2018/11/27 12:28:19 Starting Robot bot ...
2018/11/27 12:28:19 Starting connections...
2018/11/27 12:28:19 Starting connection Firmata-1FAE66071409C015 on port ./dev/ttyACM0...
2018/11/27 12:28:19 1 error occurred:
        * The system cannot find the path specified.

How to solve this problem

deadprogram, 2018-11-28:
Hello @recabyte008 when using Windows your port for connecting to the Arduino is named differently than on Linux. See https://github.com/hybridgroup/gobot/tree/master/platforms/firmata#windows for more information.

trevrosen, 2018-12-17:
Do we need further work on this issue or can I close?

closed 2019-01-02 by trevrosen