hybridgroup/gobot #644: Set QoS for MQTT Publish/Subscribe

wieringen, 2018-12-15:
It would be nice to be able to set the QoS when subscribing/publishing to events. The config stream in google cloud iot-core for instance recommends a QoS of 1.

trevrosen, 2018-12-17:
@deadprogram the QoS is currently hardcoded to `0` in the [subscription code](https://github.com/hybridgroup/gobot/blob/master/platforms/mqtt/mqtt_adaptor.go#L142) and also in the [publishing code](https://github.com/hybridgroup/gobot/blob/master/platforms/mqtt/mqtt_adaptor.go#L133). I can alter this API to allow a user to set in a variety of ways, but I'm looking to you for your opinion/direction here. For instance, variadic argument added to the function calls would be backwards compatible, but also kind of a hack. We could also add new version of the functions (not a big fan of that idea) or we could just live with a backwards-incompatible change (tho I'm not sure what that does to our semver assertions b/c I'm not certain what's covered by the versioning yet).

deadprogram, 2018-12-17:
Seems like adding a method like `SetQoS(val int)` to allow overriding this value, and keeping a sensible default, is the least intrusive. What do you think?

trevrosen, 2018-12-17:
yeah that can work as long we we're not looking for per-message granularity

deadprogram, 2018-12-17:
That works for me, since that is mostly how the current use case is setup.

wieringen, 2018-12-19:
You guys are fast! However I do think this needs to be granular for every subscription.
Take a look at this example from Google:
https://github.com/GoogleCloudPlatform/nodejs-docs-samples/blob/master/iot/mqtt_example/cloudiot_mqtt_example_nodejs.js

On line 257 you will see this:
```
// Subscribe to the /devices/{device-id}/config topic to receive config updates.
// Config updates are recommended to use QoS 1 (at least once delivery)
client.subscribe(`/devices/${argv.deviceId}/config`, {qos: 1});

// Subscribe to the /devices/{device-id}/commands/# topic to receive commands
// commands is recommended to use QoS 0 (at most once delivery)
client.subscribe(`/devices/${argv.deviceId}/commands/#`, {qos: 0});
```

trevrosen, 2018-12-19:
Yeah I tend to agree. Your only other option here would be to setup / teardown multiple clients, which would likely have other issues. 

@deadprogram I think we're down to either:

1. Variadic arg at the end of the existing function signature
2. A totally new function/interface
3. Altering the existing function signature (backwards-incompatible change)

Let me know which you prefer and I'll make it happen. I'm currently leaning toward **1** as it solves the current issue and doesn't break back compat.

wieringen, 2018-12-19:
New methods and deprecating the old ones seems reasonable?

trevrosen, 2018-12-19:
Yeah honestly it's more about how @deadprogram wants to manage the semver implications of a back-incompat change in this area of Gobot.

trevrosen, 2019-01-02:
I like it. I'll make it happen.

trevrosen, 2019-01-03:
@wieringen check it out and let me know what you think: https://github.com/hybridgroup/gobot/pull/650

PR #645 (2018-12-17): Allow setting QoS on MTT adaptor
 Fixes #644 

## What
Add an option for setting the QoS value sent with MTT messages in publish/subscribe events

## Why
Requested in #644 and also extends the value of the adaptor generally.
commit 0ea9edea (2018-12-17): Allow setting QoS on MTT adaptor

 Fixes #644
closed 2019-01-26 by trevrosen