hybridgroup/gobot #648: NewRelayDriver should allow for creating inverted relay

andrewebdev, 2018-12-31:
Currently `.On()` sets the relay state to high, and `.Off()` sets the relay state to low.

We have realays boards in one of our projects that are actually inverted.
This causes us to do things in the code like `relayDriver.On()` when we want it switched off, and of course it leads to some confusion down the line.

It may be a good idea for NewRelayDriver to allow for this use case?

deadprogram, 2019-05-31:
Seems like a good idea. A PR that added this would be gladly accepted.

andrewebdev, 2019-06-15:
@deadprogram just a ping to remind that I have done a PR for this. (just in case this issue was overlooked)

kneerunjun, 2019-10-16:
```go
func (l *RelayDriver) On() (err error) {
	if err = l.connection.DigitalWrite(l.Pin(), 1); err != nil {
		return
	}

	if l.Inverted {
		l.high = false
	} else {
		l.high = true
	}

	return
}
```
I notice you are still writing 1 to DigitalWriter (Outside the Inverted if logic) in Driver.On - this is hardly helping. Most of the relays close NO-C on LOW ! Wouldn't it help to check Inverted before you DigitalWrite 1.

stuartleeks, 2020-03-29:
As @kneerunjun noted, `On()` always writes `1` and `Off()` always write `0`. I expected `inverted` to invert the value that was written as well. Unfortunately this means that I'm still having to use `Off()` to turn on the component attached to the relay.

Would you accept a PR that takes the `Inverted` field into account when writing?

deadprogram, 2020-03-29:
Yes, please, and thanks!

stuartleeks, 2020-03-29:
Great :-)

andrewebdev, 2020-03-31:
@stuartleeks Thank for this. I saw this email yesterday and was going to start updating this today; Then I noticed you already submitted a pull request :)

Not sure how I missed that it still writes 1 instead of 0. Our code that uses this functionality is currently still doing the old manual if-check, so we never actually caught this bug in our code.

Good job catching it!

stuartleeks, 2020-03-31:
Thanks @andrewebdev. I was working on a home project with my raspberry pi and thought I'd misunderstood gobot or mis-wired my relay. After checking my circuits and code I started digging in to the gobot code and found what @kneerunjun described which led me to this issue. 

Hopefully we can get the PR merged and the change released soon, and then I can use the `Inverted` setting and remove the big `NOTE` comment I have in my code explaining the use of `Off()` for `On()` :grin:

gen2thomas, 2023-01-28:
Seems to be fixed. Otherwise please try "WithPinActiveLow()", available on "dev" branch.

PR #742 (2020-04-06): Update value written via DigitalWrite if RelayDriver.Inverted is true
The goal of this PR is to update the RelayDriver to write `0` instead of `1` for `On()` if `Inverted` has been set `true` (and similar inversion for `Off()`, as per [this comment](https://github.com/hybridgroup/gobot/issues/648#issuecomment-605701451) and [initial agreement](https://github.com/hybridgroup/gobot/issues/648#issuecomment-605701691) from @deadprogram 

I ended up making more changes in this PR than I initially expected, but they are broken into three separate commits if that helps:

* Update `gpioTestAdapter` so [... cut]
closed 2023-01-28 by gen2thomas