hybridgroup/gobot #657: Dualshock 4 still off?

csduarte, 2019-03-04:
Relates to #547 
I was recently experimenting with a few different devices. I've tried two different PS4/Dualshock4 controllers and the current mappings are way off. 

I'm on Mac OSX, 10.14.3. SDL2 v2.0.9, Gobot 1.12.0. Controllers appear to be standard Sony made ps4 controllers.

I would just file a PR, but it doesn't make sense that the mappings would change so quickly, does it? Are there other reasons the mappings could be off?

trevrosen, 2019-03-04:
There are subversions of that controller, but since you're on a Mac you might want to give the JSON config a shot. We had a macOS-related problem at the hack day during the last GopherCon and had to do this for Dualshock 3, so give the same a shot for Dualshock 4 and see if that gets you there: https://github.com/hybridgroup/gophercon-2018/blob/master/drone/tello/step5/main.go#L34

csduarte, 2019-03-05:
Thanks. I'll try it out and open up a PR if needed. Thank you.

closed 2019-03-05 by csduarte